riscv/riscv-fast-interrupt #3: Context switches?

ilg-ul, 2018-06-25:
How do you plan to handle context switches in an embedded RTOS? Via the `msip`?

kasanovic, 2018-06-28:
csip was intended to be used for this, leaving msip for interprocessor interrupts.

ilg-ul, 2018-06-28:
For this mechanism to work properly, the context switch interrupt must be guaranteed to have the lowest priority.

https://github.com/emb-riscv/specs-markdown/blob/develop/interrupts-use-cases.md#dedicated-context-switch-interrupt

In other words it should not be possible to have interrupts with lower priority, regardless the user settings.

kasanovic, 2018-06-28:
Yes, the user must set the lowest non-zero interrupt level and lowest priority for csip in this use-case.

ilg-ul, 2018-06-28:
If all interrupts are set to the lowest non-zero interrupt level and lowest priority, is `csip` guaranteed to be the last entered? Isn't the order given by th interrupt IDs?

kasanovic, 2018-06-28:
Yes, the current spec breaks ties with lowest interrupt id winning, so you'd lose one level/priority setting to create a background interrupt.

ilg-ul, 2018-06-28:
Sorry, I don't understand your phrase. Is `csid` guaranteed to be the last interrupt entered or not?

kasanovic, 2018-06-28:
No. Every interrupt input including csip has a variable level+priority setting.  The user is responsible for setting csip to have the lowest level+priority.  With current spec, if all inputs were set to same level+priority, then csip would not be the last interrupt entered, as level+priority ties are resolved in favor of lower-numbered interrupts.

ilg-ul, 2018-06-28:
Then you're simply asking for trouble.

In a good design it should be easy to do the right thing and hard to do the wrong thing. Your design does not meet this requirement.

ilg-ul, 2018-06-28:
However, given your explanation and as initially asked, `msip` is guaranteed to be the last interrupt entered, and can be safely used for context switches.

You can name `csip` as you wish, but don't recommend it for context switches.

kasanovic, 2018-07-07:
Actually, with current priority tie breaking, msip would  have highest priority if all other settings are same, as it has lowest number.  The default scheme has timers and software interrupts at bottom of interrupt id space, so hard to use interrupt id to give a certain ordering.  An option is to add config bits to set certain interrupts at extreme priorities (lower than all other, or higher than all other).

aswaterman, 2018-07-07:
Tiebreaking in favor of lower numbers matches the PLIC.  Having the CLIC and PLIC differ in this respect would be especially confusing.

ilg-ul, 2018-07-08:
> Having the CLIC and PLIC differ in this respect would be especially confusing.

Definitely.

kasanovic, 2018-07-09:
The CLIC is replacing the previous CLINT, and in CLINT the highest-numbered ID won in a tie (opposite of PLIC).   I'm changing the spec to make highest-numbered interrupts win in a tie, as this is actually more compatible with previous situation. It also has effect of making software and timer interrupts have lower priority than other interrupt inputs, which helps with the use case Liviu gives.  The two software interrupts (msip/csip) are either side of the standard timer in interrupt id ranking, so can have both default orderings of software interrupt versus timer priority.

ilg-ul, 2018-07-09:
> I'm changing the spec to make highest-numbered interrupts win in a tie

Great!

Since CLIC is actually a new design, I don't think that compatibility with CLINT is an absolute must.

I would suggest a slight reorder:

```
 0  csip        CLIC software interrupt
 1  reserved
 2  reserved
 3  reserved

 4  utip        User timer interrupt
 5  stip        Supervisor timer interrupt
 6  reserved
 7  mtip        Machine timer interrupt

 8  usip        User software Interrupt
 9  ssip        Supervisor software Interrupt
10  reserved
11  msip        Machine software interrupt

12  ueip        User external (PLIC) interrupt
13  seip        Supervisor external (PLIC) interrupt
14  reserved
15  meip        Machine external (PLIC) interrupt
```

This would guarantee `csip` to have the lowest possible priority and make it suitable for context switches, then also guarantee the timer interrupt to have lower priority then all other.

kasanovic, 2018-07-09:
msip and  csip are functionally identical in CLIC mode.  You can simply use msip for local background thread and csip for interprocessor interrupt, and get same effect as you desire without renumbering anything.

ilg-ul, 2018-07-09:
If you agree that `msip` can be used for context switches, I suggest you explicitly mention this in the document, to be clear for everybody.

closed 2018-07-09 by ilg-ul