riscv/riscv-fast-interrupt #5: Stack overflow detection?

ilg-ul, 2018-07-13:
How do you plan to handle detecting stack overflow?

I know that for the privileged profile this is not an issue, since the MMU will handle run-time stack growth, but for microcontrollers the stack overflows are probably the major cause for crashes.

FYI, the microcontroller profile proposal uses two CSRs, one for the main stack and one for the thread stack:

https://github.com/emb-riscv/specs-markdown/blob/develop/csrs.md#main-stack-pointer-limit-spmlimit

aswaterman, 2018-07-13:
Some combination of PMPs or PMAs can address this problem, right?  If you put the stack at the bottom of some RAM, then stack overflows will violate a PMA check.  PMPs can work in the other cases.

ilg-ul, 2018-07-14:
In theory yes, you can define a memory region and when you detect accesses outside it you trigger an exception, but I don't see how this mechanism can work for a microcontroller which has no MMU/MPU.

My proposal uses a much simplified mechanism, a register that is compared with the stack each time the stacjk grows. It must be saved/restored with each context switch, so it is beter reflected in one of the general registers (which are plenty).

brucehoult, 2018-07-16:
The smallest SiFive core, the E20, has in its standard configuration four regions of Physical Memory Protection. Using one of those for the stack would make perfect sense. If you want a custom core, you can increase that up to eight regions, or decrease it to one or even zero. That would save a little bit of hardware size, but if you want to detect stack overflow I'd suggest having at least one region of PMP would be the lowest overhead way to do it.

Doing it in software bloats the program size and slows the execution speed, and the checks have to be duplicated in *every* function (or have a call from every function to some millicode routine). Enabling a single PMP region has got to be cheaper in aggregate.

ilg-ul, 2018-07-16:
> The smallest SiFive core, the E20, has in its standard configuration four regions of Physical Memory Protection. 

Version v1p0 of the SiFive E20 Core Complex Manual states:

"... PMP with None x regions and a minimum granularity of None bytes."

You probably mean the slightly larger device, E21, which indeed has "... PMP with 4 x regions and a minimum granularity of 4 bytes."

> ... if you want to detect stack overflow I'd suggest having at least one region of PMP would be the lowest overhead way to do it. 

I checked again the privileged PMP specs (chapter 3.6) and I cannot figure out how to configure 
PMP for detecting stack overflow; I'm afraid this is a misunderstanding and the requirements for stack overflow detection need to be clarified.

---

A typical microcontroller RTOS application does not have a single stack, but many, one for interrupts and one for each thread.

Each thread stack can have any size and can be allocated at any legal RAM address at any time (most probably aligned to some reasonable limit, like 8 bytes, as provided by malloc()). The number of thread stacks is limited only by the amount of RAM (actually _fast_ RAM) available. Threads can be created and destroyed at any time and in any order.

Given that the thread stack can be located anywhere, the only way to describe the memory region using the privileged specs PMP, is with a pair of TOR configurations, one for the beginning and one for the end address. (Please note that NAPOT configurations are generally uselles for microcontroller devices, where memory is tight). The interrupt stack would probably need another pair of TOR configurations, so 4 configurations are already required for the two stacks.

As a side note, these definitions alone would add 5 more registers (the `pmpcfg0` and 4 `pmpaddr_i_` addresses) to the context switch overhead.

However, given that any memory words located immediately below the stack are also at legal RAM addresses, and can be anything (variables, bufers, other stacks, etc), acesses to them via regular r/w accesses should not be prevented.

So, as long as the PMP cannot distinguish stack accesses from regular accesses, how can it prevent the stack to overwrite those words below it?

---

As a personal experience, although memory protection units may seem like a nice feature to have, for typical microcontrollers using today RTOSes, they are of little practical value, if any. RISC-V PMP seems no exception.

brucehoult, 2018-07-17:
Sorry, I misspoke. I meant "E2 Series Core", which is configurable as you desire, not "E20", which is one particular "standard" configuration. Hopefully that should have been obvious, given that I was talking about configuration options, but sorry for the confusion.

As for the issue with lots of threads and thread stacks and the overhead of thread switching, you're faced with the question: which is more common, function calls or context switches? If function calls are significantly more common then use hardware protection. If context switches are significantly more common then use software checks of stack bounds. It's a simple matter of optimisation for your own workload, not a religious question that can be answered once and for all for everyone.

Another much lower overhead option is to prove that your stacks can't overflow. If you avoid recursive functions (as I expect you do) and alloca() and calls through unbounded function pointers then it's a very simple matter to analyse your code's call graph and calculate the maximum possible stack usage.

I've seen a number of processor-specific utilities for this around (e.g. AVR) but this one looks to be processor-independent and requires only that you use a reasonably modern gcc: https://github.com/PeterMcKinnis/WorstCaseStack

Well, you probably already have your own favourite utility for this job.

ilg-ul, 2018-07-17:
The question was obviously related to hardware protection; I renamed the subject to reflect this.

> you probably already have your own favourite utility for this job.

unfortunately there is no such magic tool, and this is why I insist on a hardware solution.

software checks at each function call, as you mentioned in the first message, 'bloats the program size and slows the execution speed', so are unacceptable.

using the RISC-V PMP, as defined by the current privileged specs, is also not realistic.

I personally know two possible solutions:

- the direct and simple one is with hardware comparators; each time a read/write access is done indirectly via SP, the address is checked; if lower than the limit, a trap is generated; the limit register is initialised at 0, and must be saved at each context switch; this was for the thread SP; a second comparator is needed for the interrupt stack.
- an enhanced MPU, which can differentiate between regular read/write accesses and stack accesses; for example the region configuration may include a bit to tell this (and in this case a fixed register is checked), or even a field where the number of the register is configured, and when accesses are done indirectly via this register, the address is validated.

For microcontrollers, the second option is probably too heavy, and I cannot recommend it.

kasanovic, 2018-07-20:
Dedicated stack overflow protection registers might be useful, but this is orthogonal topic to interrupts.

closed 2018-07-20 by kasanovic

ilg-ul, 2018-07-20:
> this is orthogonal topic to interrupts

Interrupts are only part of the problem, and the CLIC does not make for a microcontroller profile. Interrupts come with stack contexts, handling multiple stacks during context switches, and exceptions thrown by stack protection mechanisms. All these should be considered together.

I can reopen the issue in the ISA repo, but you'll probably abruptly close it there too.