riscv/riscv-fast-interrupt #6: Priority threshold?

ilg-ul, 2018-07-14:
How do you plan to handle disabling interrupts below a priority threshold?

In the microcontroller profile proposal I used a dedicated CSR:

https://github.com/emb-riscv/specs-markdown/blob/develop/csrs.md#interrupt-priority-threshold-iprioth

kasanovic, 2018-07-20:
I have been thinking about how to provide this feature.  It will add to, and interact with, existing context state, and with retrieving next valid interrupt.

dansmathers, 2020-04-23:
xintthresh CSR added june 10 2019 clic.adoc.

From draft-20200423:

4.10. New Interrupt-Level Threshold (xintthresh) CSRs
The interrupt-level threshold (xintthresh) is a new read-write CSR, which holds an 8-bit field (th) for the threshold level of the associated privilege mode.

A typical usage of the interrupt-level threshold is for implementing critical sections. The current handler can temporarily raise its effective interrupt level to implement a critical section among a subset of levels, while still allowing higher interrupt levels to preempt.

The current hart’s effective interrupt level would then be: effective_level = max( xintstatus.xil, xintthresh.th )

closed 2020-05-12 by Kevin-Andes