riscv/riscv-fast-interrupt #7: Interrupt priorities vs. (too few) preemptable interrupt levels?

ilg-ul, 2018-07-16:
If I understand the specs right, each of the 1024 interrupts has a configurable priority, stored as a  `clicintcfg[i]` byte.

This byte seems to encode mode, level and priority; unfortunately I'm not able to understand the use of the 'mode' bits for M or M/U microcontrollers, but the specs clearly define that there are no more than 16 levels which correspond to no more than 16 different preemption values.

The rest of the bits in the priority bytes are used to order simultaneous interrupts, but without causing preemption.

I don't know the use cases you considered with this design, but for microcontroller usage, this practically translates into no more than 16 usable priorities, which, for more complex applications, might be a serious limitation.

FYI, regardless how many distinct interrupt priority levels exist, microcontroller developers are used to expect that a higher priority interrupt will preempt a lower priority interrupt. For a byte priority field, 256 preemptable levels are expected.

The 8-bit limit for the entire priority field is also debatable, and too low in my opinion. (In the microcontroller profile proposal this limit is not set in the architecture specs, but left to the silicone vendor implementation, which might be a more reasonable aproach).

kasanovic, 2018-07-20:
While some systems might support a large number of preemption levels, my interaction with architects at very large micro controller manufacturers suggest they think when customers are using more than a handful of levels, they don't really understand their application well.   Preemption, even with full hardware support, has a high throughput/energy cost per event, and users should try to restructure application code to have less nested preemption and more non-preemptive scheduling.  Can you give an application example where more than 15 preemption levels make sense?

As for total number of priority bits, customers are very sensitive to silicon cost, and each extra bit/input is another flop in design.  Usually customers prune this number down to save area.   There's also an advantage to keeping the memory aperture within a single 4K page (protection, reach with a 12b offset, etc.).  That said, it would be possible to expand priority storage later in a compatible way by adding a second larger memory aperture.

ilg-ul, 2018-07-20:
> manufacturers ...  customers don't really understand their application well ... restructure application code to have less nested preemption

it's good that now manufacturers teach us how to structure our applications.

if you are concerned with throughput/energy, you can make a notice in the manual, and/or set the default to 16, but normally the developers should have the final say, and if they think that all priorityes should preempt, they should be allowed to configure the device to do so.

you are simply introducing an artificial limitation.

ilg-ul, 2018-07-21:
to be sure I am not misunderstood, the complaint is not against differentiating preemption levels from priorities, the mechanism itself is probably ok, and for some applications it may be useful; I'm only suggesting to allow more preemption levels.

kasanovic, 2018-08-31:
We agreed to up the number of levels of preemption levels to 256, as with current design the hardware cost is very low.  (I still doubt many, if any, customers will pay for more than four bits of interrupt priority per input - but am happy to learn about any silicon that implements this).

kasanovic, 2018-08-31:
Updated spec to support all 8 bits being used as preemption levels.

closed 2018-08-31 by kasanovic