riscv/riscv-fast-interrupt #10: Expand some terse commentary

nwf, 2018-08-30:
In 6.1, the prose reads  "or mscratch can be used to point to more extensive thread-local context that might contain a stack pointer".  This dance is trickier than this text makes it sound, or, at least, less generally applicable: at the entry to a vertical interrupt service handler, there may be no actual scratch registers, the name of `mscratch` notwithstanding.  The entire register file (save the instruction pointer) is state of the lower permission level, and `mscratch` is, by assumption, a pointer to the thread-local context.  The only safe place to spill registers, then, is at some fixed offset within this context, as that is all that can be reached after swapping `mscratch` with a general purpose register.  Thus, either the thread-local context is adjacent to the stack or, if it points at a separate stack, it must also contain a "mini stack" to store enough context to load sp while neither trashing the context pointer nor any register state.  A second scratch register would allow an ISR to follow a chain of pointers before spilling anything, and so could permit a single stack pointed to by a context value in `mscratch`.  Since the proposed CLIC spec is already adding several CSRs, perhaps a second scratch CSR for each of M/S/U, a la MIPS, could be on the table?  (Speaking of such things, I am made vaguely uncomfortable by the blurred lines between the "privileged spec" and the CLINT and CLIC specs, especially as pertains to CSRs.  Is there an intention of either combining or more completely decoupling these, going forward?)

Having just described the difference between horizontal and vertical interrupt transfers and the usual `mscratch` dance in 6.1, the prose proceeds, in 6.2, to give code that is only safe for horizontal transfers with a meaningful sp register.  I think the listings in 6.2, 8, 9.1, and 9.2 should have disclaimers at the top a la the listing in 7.1.

10.1 asserts "This code can be used in a model where user-level code has one stack, and all interrupts and exceptions are handled on a second M-mode-only stack" and so should probably warn that *not* doing this stack split represents a likely information leak in the best case or a privilege escalation vector (in a multi-hart system with shared pages for U-mode stacks).

Given the desire for low-latency interrupt handling, why are timers forced through M mode in M/S/U (and M/U) systems?  A `stimecmp` word would be nice to have, I think, and would remove (some of) the need for a SBI.  Similarly, the ability to IPI directly into S mode on another core would also be nice, by exposing harts `ssip` bits, perhaps as the 2 bit in the existing `msip` fields, which would also, then, be visible in the S-mode memory region.

P.S. Some nits:

In 2.2, "but can also masks any" should just be "but also masks", as there does not seem to be anything modal about this behavior?

Section 4 (and 13) refer to a `cliccfgint` which should probably be `clicintcfg`?

kasanovic, 2018-08-31:
Please resubmit as multiple issues.

closed 2018-08-31 by kasanovic

nwf, 2018-09-01:
Done.  #11, #12, #13, #14, #15