riscv/riscv-fast-interrupt #11: Flesh out the "thread-local context" mscratch story beyond the terse description given

nwf, 2018-09-01:
In 6.1, the prose reads "or mscratch can be used to point to more extensive thread-local context that might contain a stack pointer". This dance is trickier than this text makes it sound, or, at least, less generally applicable: at the entry to a vertical interrupt service handler, there may be no actual scratch registers, the name of mscratch notwithstanding. The entire register file (save the instruction pointer) is state of the lower permission level, and mscratch is, by assumption, a pointer to the thread-local context. The only safe place to spill registers, then, is at some fixed offset within this context, as that is all that can be reached after swapping mscratch with a general purpose register. Thus, either the thread-local context is adjacent to the stack or, if it points at a separate stack, it must also contain a "mini stack" to store enough context to load sp while neither trashing the context pointer nor any register state. A second scratch register would allow an ISR to follow a chain of pointers before spilling anything, and so could permit a single stack pointed to by a context value in mscratch. Since the proposed CLIC spec is already adding several CSRs, perhaps a second scratch CSR for each of M/S/U, a la MIPS, could be on the table? (Speaking of such things, I am made vaguely uncomfortable by the blurred lines between the "privileged spec" and the CLINT and CLIC specs, especially as pertains to CSRs. Is there an intention of either combining or more completely decoupling these, going forward?)

kasanovic, 2018-09-01:
Additional scratch CSRs would add overhead to every CPU, and the current scheme has been shown to work fine for multiple RTOS & Unix ports, so there is no clear need.  At best, each scratch CSR saves four instructions (save/swap/.../swap/load) that don't cause additional cache misses (OK  -maybe one more I-cache miss at most) in the infrequently run and much longer context-swap sequence.

Why does the thread-local context have to be adjacent to the stack?  Once some registers are saved in the TCB, then the code can arbitrarily interrogate the TCB.

closed 2018-09-01 by kasanovic

nwf, 2018-09-01:
I was presenting an either/or, not an absolute: either the TCB is next to the stack so you can spill to the stack using a fixed offset from the TCB pointer *or* the TCB contains a spill space and some registers don't end up on the stack at first.  The first is convenient for OS entry models that want a unified trap frame, as it avoids the need to shuffle saved state over to the stack later.  The second is advantageous if, for some reason, stacks and contexts are to be divorced.  I think we are in agreement of the actual details, regardless.