riscv/riscv-fast-interrupt #16: minhv

dangerblue, 2018-09-18:
hi，the minhv are used for retr

Kevin-Andes, 2018-11-01:
You have very good observation and indeed this is a problem.  In current spec., mepc and mcause will always be overwritten whenever an exception or NMI occur. Therefore, we are proposing a general solution to solve this problem fundamentally. 

Our basic idea is to have 2 CSRs to automatically backup (stack) mepc and mcause so that exceptions/NMI can always be recovered 100%. We will discuss more details in the meeting.

kasanovic, 2018-11-26:
We decided that the separate NMI approach is actually orthogonal to the local interrupt controller, so the additional NMI support is now being proposed as extension of main privileged architecture. Closing here for this reason.

closed 2018-11-26 by kasanovic