riscv/riscv-fast-interrupt #26: More on xINHV

marceg, 2018-12-01:
An issue was already raised highlighting that exceptions on interrupt handler lookup reads are non-recoverable.  I see the issue was closed, but the spec appears to still contain the issue.  Some further observations:

* There is no value in re-attempting the table read on return from the exception.  If an interrupt is still pending, it will be taken normally, triggering another table read anyway.  And if a higher priority interrupt is now pending, it is preferable to take that higher priority interrupt in any case.
* The normal exception mechanism already sets xCAUSE.xTVAL to the faulting address, and here xEPC needs to point to the PC of the next instruction to execute in the interrupted code as normally expected.
* The spec unnecessarily defines micro-architectural behavior of xINHV, such as its value while attempting the table read, which is not visible to software and thus outside the scope of an architectural spec.  Documenting only architectural behavior is simpler and clearer.  In this case, as far as I can tell, the only point at which xINHV is architecturally relevant is in the value set in xCAUSE when an exception is taken due to a failed read of the handler table.

Perhaps the name of xINHV can be generalized as an indication that the exception has occurred because of an asynchronous event, and not because of the instruction stream pointed to by xEPC?

(Sidenote.  Of course, allowing handler table lookups to take a page fault is generally not a good idea.  There are several reasons one may want to pin all pages used by handlers.  That's a whole other, mostly software topic.)

dansmathers, 2020-04-23:
Update: no significant change to xinhv documentation between Nov25 2018 and draft 20200423: 
Here is the wording:
 
In this case, the processor switches to the handler’s privilege mode and sets the hardware vectoring bit xinhv in xcause, then fetches an XLEN-bit handler address from the in-memory table whose base address (TBASE) is in xtvt. The trap handler function address is fetched from TBASE+XLEN/8*exccode. If the fetch is successful, the processor clears the low bit of the handler address, sets the PC to this handler address, then clears the xinhv bit in xcause. The overall effect is:

pc := M[TBASE + XLEN/8 * exccode] & ~1

Note | The hardware vectoring bit xinhv is provided to allow resumable traps on fetches to the trap vector table.

kasanovic, 2021-04-13:
Closed by PRs #134 and #131

closed 2021-04-13 by kasanovic