riscv/riscv-fast-interrupt #29: Interrupt trigger in the trigger module

chuanhua, 2018-12-11:
In version 0.13 of the RISC-V debug spec, there is an itrigger (interrupt trigger) in the trigger module defined based on _mie_ CSR. For CLIC, _mie_ CSR is not used. I think CLIC spec should also define the corresponding change in the itrigger mechanism.

Propose to use _tdata2_ CSR to trigger on 2 interrupt IDs for RV32 or 5 interrupt IDs for RV64.

chuanhua, 2018-12-12:
To be more specific, propose the itrigger scheme as follows.

- **RV32**: two interrupt IDs
```
tdata2 [0]     : interrupt 1 ID valid
tdata2 [12:1]  : interrupt 1 ID
tdata2 [16]    : interrupt 2 ID valid
tdata2 [28:17] : interrupt 2 ID
```

- **RV64**: Four interrupt IDs
```
tdata2 [0]     : interrupt 1 ID valid
tdata2 [12:1]  : interrupt 1 ID
tdata2 [16]    : interrupt 2 ID valid
tdata2 [28:17] : interrupt 2 ID
tdata2 [32]    : interrupt 3 ID valid
tdata2 [44:33] : interrupt 3 ID
tdata2 [48]    : interrupt 4 ID valid
tdata2 [60:49] : interrupt 4 ID
```

dansmathers, 2020-05-12:
kevin wrote to discussion group 5/7/2020:
Hi,
In the current Debug Spec, a trigger module is defined and it can take interrupts as triggers (Interrupt Triggers). Triggers are useful because they can cause a breakpoint exception, entry into Debug Mode, or a trace action. In addition, triggers can function without executing instructions or even without the Debug Module.

However, Interrupt Trigger (itrigger) is currently only defined for interrupts configurable in mie. Therefore, we need to extend it to cover CLIC. After some brief discussion with Tim (the chair of the Debug Task Group), he suggested that we should implement CLIC itrigger in a way such that it does not cause large changes to the Debug Spec. More specifically, Tim suggested: “have the CLIC generate a single debug interrupt signal, which connects to each hart, where itrigger contains a single bit indicating whether to pay attention to the signal or not. Then the definition of exactly what set of interrupts can be selected to cause a debug interrupt can be in the CLIC spec and not in the Debug Spec.”

Based on Tim’s suggestion, I would like to propose the following simple scheme for CLIC interrupt triggers (which slightly extends Chuanhua’s original proposal):

CLIC can optionally support from 0 up to 32 interrupts as triggers. These triggers are specified in optional 16 memory-mapped WARL registers for each privilege mode. To program them, users first need to check their availability by writing one’s to these registers and reading the values back.

M-mode CLIC memory map

Offset
0x0000   1B   RW cliccfg
0x0004   4B   R  clicinfo
…
0x0040   4B   RW clicinttrig[0]
0x0044   4B   RW clicinttrig[1]
0x0048   4B   RW clicinttrig[2]
0x004C   4B   RW clicinttrig[3]
0x0050   4B   RW clicinttrig[4]
0x0054   4B   RW clicinttrig[5]
…
0x0074   4B   RW clicinttrig[13]
0x0078   4B   RW clicinttrig[14]
0x007C   4B   RW clicinttrig[15]
…

Each clicinttrig[i] register has the following fields to program and enable a particular interrupt as a trigger. This layout also reserves some room for extension (if we ever want to increase the maximum number of CLIC interrupts in the future).

bits     field
31:29    reserved (WPRI 0)
28:17    ID for interrupt 2*i+1
16        enable for interrupt 2*i+1
15:13    reserved (WPRI 0)
12:1     ID for interrupt 2*i
0         enable for interrupt 2*i

kasanovic, 2020-05-12:
Suggest we put interrupt enable in bit 15 of each field, and put interrupt id in bits 11:0, with bits 14:12 reserved for expansion.

kasanovic, 2020-05-12:
TG suggests to pack as only one interrupt per 32-bit word

kasanovic, 2020-05-12:
Need to confer with debug task group regarding pending versus taken interrupts as triggers.

Kevin-Andes, 2020-06-02:
After discussing with the chair of the Debug Task Group, he said CLIC should support the existing standard triggering behavior but it is fine for CLIC to create additional triggering conditions.

However, later in our group meeting, we decided that existing interrupt triggering conditions should fit our needs in the current stage, so we will not create additional triggering conditions such as “upon pending” or “pending and enabled.”

Therefore, the triggering condition according to the current Debug Spec is: “the trigger only fires if the hart takes a trap because of the interrupt.“

Also, when the trigger fires, “the requested action is taken just before the first instruction of the trap handler is executed.“

kasanovic, 2021-03-16:
The memory map table should have the reserved section reduced to omit the region allocated to interrupt triggers.

Kevin-Andes, 2021-03-30:
Updated the reserved section with commit 8496573b01db41b62fefba4863a1822a1309b9ea.

kasanovic, 2022-06-07:
Closed by https://github.com/riscv/riscv-fast-interrupt/pull/190

closed 2022-06-07 by kasanovic