riscv/riscv-fast-interrupt #30: mcause format for exceptions under CLIC mode

chuanhua, 2019-01-17:
It is not clear from the CLIC proposal that when an **exception** happens under CLIC mode, the mcause CSR format for recording the exception reason is the same as the **original** format from the privileged architecture spec or is the same as the **new** CLIC mcause format.

Could you clarify on this point?

billhuffman, 2019-02-26:
- I think previous interrupt level isn't needed for exceptions because it doesn't change.  Maybe it doesn't hurt, but maybe it does.  Do we actually want it restored when we return from the exception?  Does the tie-in between exceptions and interrupts cause any "interesting" behavior?  If exceptions and interrupts are on different stacks?
- xPP and xPIE are needed somewhere, but maybe not in xcause because xstatus is saved anyway.
- xinhv seems like it IS needed for exceptions.  Perhaps not for interrupts as interrupts won't be enabled until cause has been saved after an exception during hardware vectoring.

The spec does need more clarity to be clear what it actually says.

Kevin-Andes, 2020-05-26:
For exception, in CLIC mode, mcause has "CLIC format." In other modes, it has the original format.

Kevin-Andes, 2020-06-11:
Clarified in the spec with commit e0f428d.

closed 2020-06-11 by Kevin-Andes