riscv/riscv-fast-interrupt #31: Does Interrupt Level Affect WFI?

billhuffman, 2019-02-26:
I assume from the sense of when WFI falls through in the privilege spec, that under CLIC, it falls through even when its priority is below the interrupt level in intstatus as that seems to be a global interrupt "enable" of sorts.  It would be good to have a sentence saying that for CLIC - if it's true.

kasanovic, 2019-09-24:
Needs more clarification please?

kasanovic, 2019-09-24:
In general, WFI can be a NOP so implementation does not have to hold WFI If interrupt is below threshold.

kasanovic, 2019-09-24:
Consensus was that the implementation did not have to exit WFI if threshold was above pending interrupt's level.  This is consistent with threshold  on PLIC, which filters interrupt before being received at processors.

JamesKenneyImperas, 2020-04-15:
I am also interested to know what precisely causes a WFI wakeup event. This should be stated explicitly.

aswaterman, 2020-04-15:
I think this is pretty clear in the priv spec already: any interrupt that's unmasked (regardless of current global mask) is guaranteed to wake up the hart, and implementations are additionally free to wake up for any other reason.

JamesKenneyImperas, 2020-04-15:
I think I still need to thrash out the details a bit here...

The CLIC has a potentially large number of inputs, each with its own enable and pending field. For these, it selects the highest priority pending-and-enabled interrupt and presents it to the core. The presentation of any interrupt in this manner is therefore a WFI wakeup event. Is that right?

I am still trying to understand what exactly _pending-and-enabled_ means in this context (see issue #52). Originally I thought that it must be the interrupt-specific enable bit _and_ the `mstatus.xie` bit, but I think now that this cannot be right, because the WFI wakeup event must occur even when `mstatus.xie` is 0. However, if the presented interrupt is _not_ affected by `mstatus.xie`, it seems that under certain circumstances interrupts may be blocked from the core when they should not be. For example, suppose the the CLIC and core are configured in this way:

1. Interrupt A is M-mode, pending and enabled in the CLIC;
2. Interrupt B is S-mode, pending and enabled in the CLIC;
3. `mstatus.mie`=0
4. `mstatus.sie`=1

My questions are:

1. Does the above setting cause a WFI wakeup event?
2. Is the pending-and-enabled S-mode interrupt taken, or is it blocked by the `mstatus.mie`-disabled M-mode interrupt? In other words, do the `mstatus.xie` bits affect the presented interrupt, or not?

Thanks.

aswaterman, 2020-04-15:
> I think I still need to thrash out the details a bit here...
>
> The CLIC has a potentially large number of inputs, each with its own enable and pending field. For these, it selects the highest [...]

AFAICS, this paragraph in the priv spec answers both questions: https://github.com/riscv/riscv-isa-manual/blob/baea02550976abe89db299bfff01e9ef82b269a3/src/machine.tex#L2466

> 
> Thanks.

JamesKenneyImperas, 2020-04-15:
I agree this clarifies WFI (not affected by `mstatus.xie`) - but what about the interrupt presented by the CLIC? Is this the M-mode interrupt or the S-mode one?

billhuffman, 2020-04-15:
Once the wakeup has occurred, the taking of the interrupt (or not) is the same as any other situation.  In the case you outline, it depends on the current privilege mode.  If the current privilege is M-mode, no interrupt is taken.  If S-mode or lower, interrupt B is taken.

      Bill

On 4/15/20 2:56 AM, JamesKenneyImperas wrote:
EXTERNAL MAIL

I agree this clarifies WFI (not affected by mstatus.xie) - but what about the interrupt presented by the CLIC? Is this the M-mode interrupt or the S-mode one?

billhuffman, 2020-04-15:
I spoke too quickly.  It's interrupt A (not B) that's taken at S-mode or lower privilege.

     Bill

On 4/15/20 8:25 AM, Bill Huffman wrote:

Once the wakeup has occurred, the taking of the interrupt (or not) is the same as any other situation.  In the case you outline, it depends on the current privilege mode.  If the current privilege is M-mode, no interrupt is taken.  If S-mode or lower, interrupt B is taken.

      Bill

On 4/15/20 2:56 AM, JamesKenneyImperas wrote:
EXTERNAL MAIL

I agree this clarifies WFI (not affected by mstatus.xie) - but what about the interrupt presented by the CLIC? Is this the M-mode interrupt or the S-mode one?

JamesKenneyImperas, 2020-04-15:
Just to be absolutely precise here (he says, flogging the dead horse) you are saying that the single interrupt presented to the core by the CLIC is the highest-priority one that is enabled by _both_ the interrupt-specific `clicintie` register _and_ the appropriate `mstatus.xie` bit, but wakeup from WFI is gated only by the interrupt-specific `clicintie` register but _not_ the `mstatus.xie` bit. Is that right?

billhuffman, 2020-04-16:
My understanding is that the wakeup and the taking of the interrupt are entirely separate.  The taking of an interrupt after waking from a WFI is _exactly_ like the taking of an interrupt at any other time.  So the only subject of discussion about WFI is the wakeup.

First, it should be stated that WFI is never forbidden to wake.  On any given cycle it is allowed to wake.  The issues are when it _must_ wake and when it is most efficient for it not to wake.  For most issues, the current privileged spec is quite clear.  It says:   "The WFI instruction can also be executed when interrupts are disabled. The operation of WFI must be unaffected by the global interrupt bits in mstatus (MIE and SIE) and the delegation register mideleg (i.e., the hart must resume if a locally enabled interrupt becomes pending, even if it has been delegated to a less-privileged mode), but should honor the individual interrupt enables (e.g, MTIE) (i.e., implementations should avoid resuming the hart if the interrupt is pending but not individually enabled). WFI is also required to resume execution for locally enabled interrupts pending at any privilege level, regardless of the global interrupt enable at each privilege level."

So the relationship of waking to privilege, global interrupt enables, and local interrupt enables is clear.  Note that even if an interrupt is for a privilege lower than the current privilege, wakeup must occur.

What I haven't seen stated clearly in the CLIC spec is whether an interrupt which is masked by the current interrupt level (xintstatus) or by the interrupt threshold register (xintthresh) must wake from WFI.   I assume from the fact that an interrupt for a lower privilege must wake that an interrupt masked by xintstatus or xintthresh must wake.  It's hard to imagine an interrupt causing wakeup for a lower privilege but not for the same privilege but lower priority.  But I believe the CLIC spec should say explicitly.

Kevin-Andes, 2020-05-26:
Should be defined when WFI is allowed to stay asleep.

kasanovic, 2020-05-26:
The global interrupt enables in xstatus.xie determine whether a pending-and-enabled interrupt causes a trap (jump to trap handler).  Any pending-and-enabled interrupt must cause WFI to be exited (whether or not  xstatus.xie is set).

Individual interrupt has to be enabled by individual enable bits to be taken. If pending interrupt is in higher-privilege mode, then higher-privileged xie bit is ignored (always enabled).  WFI does not have to retire for an otherwise enabled and pending interrupt for current privilege mode with interrupt level below the CLIC threshold for the current privilege mode.  For interrupts pending-and-enabled for a  higher privilege than current mode, the higher-privilege's CLIC threshold is ignored and treated as if set to lowest value for purpose of determining whether to take the interrupt, i.e., trap will be taken into higher-privilege mode, even if higher-privilege global interrupt enable is 0.

If global interrupt bit for current privilege in mstatus is only thing preventing trap from being taken, then WFI must wake up.  

Threshold only applies to current privilege mode.

Define pending-and-globally-enabled.

billhuffman, 2020-05-26:
I'd like to avoid repeating most of the interrupt enable definition here.

Can we say something like:  The WFI wake logic watches interrupts as if the current privilege's interrupt enable in mstatus were set, regardless of its actual current value.  If, under that condition, no interrupt would be taken, WFI is permitted to remain asleep.  If WFI wakes but no interrupt actually takes a trap, execution continues normally with the next instruction.

dansmathers, 2020-05-26:
How about this wording?  It also tries to avoid repeating the interrupt enable def.  I tried to copy wording style from the privilege spec:

In CLIC, similar to original mode, WFI is unaffected by the global interrupt bits in mstatus but should honor clicintie, xintstatus, xintthresh, i.e., events that would have caused an interrupt to be taken but are not due to the global interrupt bit settings in mstatus are required to cause WFI to resume execution.

billhuffman, 2020-05-26:
I agree it avoids repeating the definition.  It misses addressing interrupts that are to be taken, I think.

In CLIC, similar to original mode, WFI is unaffected by the global interrupt bits in mstatus but should honor clicintie, xintstatus, xintthresh, i.e., events that should cause an interrupt to be taken as well as those that would have caused an interrupt to be taken but do not due to the global interrupt bit settings in mstatus are required to cause WFI to resume execution.

OR

In CLIC, similar to original mode, WFI is unaffected by the global interrupt bits in mstatus but should honor clicintie, xintstatus, xintthresh, i.e., WFI may remain sleeping only when no interrupts are pending that would be taken if the current privilege's mstatus enable were set.

dansmathers, 2021-01-07:
Hi Kevin, shouldn't this wording
"WFI will wake up only if an interrupt"
be 
"WFI will wake up if an interrupt"  because a WFI can be a NOP and not actually sleep, right?   We just specify when it must wake up, but it can wake up earlier if it wants.

billhuffman, 2021-01-08:
Kevin and Dan,

I agree that we only specify when a WFI must wake up.  It can wake up any time it wants to in addition.  So, it sounds like Dan’s proposed wording should be better.

      Bill

EXTERNAL MAIL

Kevin-Andes, 2021-01-19:
Thank you for your valuable suggestion and I have modified the spec accordingly with the following checkin:
    c521e6f71d1f37b10dba30b06a54a63ddc28c549

Kevin-Andes, 2021-01-19:
While updating the spec, I noticed the following problems:

In spec we described: In CLIC, similar to original mode, Wait for Interrupt instruction (WFI) is unaffected by the global interrupt-enable bits in mstatus.xie but should honor clicintie[i] and xintthresh. Also, the threshold is valid only for the current privilege mode (while ignored for other privilege
modes).

However, this definition creates counter-intuitive cases. For example:

- When WFI occurs in M-mode, a pending **M-mode interrupt** level = 3, mintthresh = 4, WFI **won’t wake up** (because it is masked by m-mode threshold)

However, 

- When WFI occurs in M-mode, if there is a **U-mode interrupt** with similar level = 3, and uintthresh = 4, WFI **will wake up** in this case (because threshold is only valid for the current privilege mode (M-mode) and ignored for U-mode interrupts) 

This seems really strange because a lower privilege interrupt can wake WFI but a current higher privilege mode interrupt cannot.

Another question is: if level == threshold, should WFI wake up? In normal usage, we only care about preemption and preemption will not occur in this "equal" case. However, it is not clear whether we should wake up WFI in this "equal" case...

billhuffman, 2021-01-28:
I think if WFI occurs in M-mode that U-mode (or S-mode) interrupts are not required to cause a wake-up.  My understanding is that interrupts which would have been taken if the appropriate global enable were set (and we weren't in the WFI state) must cause a wake-up.  The computation for required wake-up, then, is the same one as for taking an interrupt except for the global enable and the WFI state.  So interrupts at lower privilege are never required to cause a wake-up.  Interrupts at a higher privilege are required to cause a wake-up without reference to any threshold.  That's the sense in which the threshold is only valid in the current privilege mode - just as it is when taking an interrupt outside of WFI.

The math comparing the pending interrupt level to the threshold should also be the same as for taking an interrupt.  So wake-up is not required to occur in the "equal" case.  (Of course, wake-up is allowed to occur at any time without a reason.)

Kevin-Andes, 2021-02-02:
In the discussion of our Task Group meeting on 20210119, Krste thought this counter-intuitive behavior was not a bug and hence acceptable. So we would not change the definition but only explain in the spec that sometimes lower privilege-mode interrupt can wake up WFI but a higher-privilege mode interrupts cannot.

Regarding whether WFI should wake up when level == threshold, we concluded that we would define it in a way such that hardware cost is minimized. Therefore, we would re-use the existing circuit for normal usage, which is detecting preemption when level > threshold. Similarly, we would change the definition to allow WFI wake up when an interrupt level > threshold (but not for == case). 

Lastly, we would also make a note in the spec to remind users that WFI should not be placed inside an ISR.

Kevin-Andes, 2021-02-02:
Personally, to avoid creating counter-intuitive cases, I agree with Bill that "interrupts at lower privilege are never required to cause a wake-up."  Maybe we can re-discuss this issue again in details in the coming group meeting (since did not have much time in previous discussion).  

I simply recorded the minutes from 20210119 but it is by no means final yet.

kasanovic, 2021-02-02:
Discussion in TG meeting was to consider dropping all thresholds from consideration for determining "locally enabled".  This provides a more consistent view of WFI wakeup, but does remove ability to use threshold to remove events from set that will wake up WFI.

Separately, another extension could be to add some way to filter lower privilege (and possibly same privilege, below-threshold interrupts) from causing WFIs (this would be advisory as WFI can always wake at any point in time).
Proposal, is uie=0, don't wake WFI on user interrupts?  Change from priv spec for CLINT.
If all lower-privilege xie=0, then can use threshold value for current privilege mode to filter wakeup events?

Kevin-Andes, 2021-02-04:
To avoid redefining the behavior of xie, here is an easy and perhaps more elegant proposal:

- If thereshold is set to "-1" (minus one, which is a meaningless/illegal value), then all thresholds are ignored. [For first usage mode]

- Otherwise, if threshold has a meaningful/legal value (0-255), then this threshold is used to mask interrupts in current privilege mode (the ones below or equal to threshold) and all lower privilege modes.  [For second usage mode]

Also, this proposal has quite low cost because we only need to add one sign bit to the threshold field to recognize minus value.

billhuffman, 2021-02-04:
Hi Kevin,

I think I understand that you’re differentiating between two modes:

  *   If the current privilege’s threshold is -1, then WFI must wake up if any interrupt whose individual enable is set occurs at any privilege.
  *   If the current privilege’s threshold is 0-255, then WFI must wake up if any interrupt occurs at a higher privilege or at the current privilege and above the threshold.  Interrupts at lower privilege or at the current privilege and at or below threshold are permitted to be ignored.

The xie bits never matter to when wake up is required.  If threshold is set to -1, that is equivalent to 0 for purposes of whether an interrupt is taken.

Is that right?   What about xintstatus?

      Bill

EXTERNAL MAIL

Kevin-Andes, 2021-02-09:
Hi Bill,

Yes, that is exactly what I meant. Thank you for clarifying and summarizing my proposal further.

My intention is to support the 2 modes we discussed (i.e., no filtering and filtering with threshold) while avoid redefining _xie_ to maintain backward-compatibility.  This can be achieved by simply adding one sign bit to the threshold.

Also, there should be no change in the behavior of _xinstatus_. It should behave just like normal. Do you notice any problem with _xinstatus_?

Thanks.
      Regards,
      Kevin

billhuffman, 2021-02-09:
The question about xintstatus was whether it affects whether a WFI wakes up.  I think the answer is "no."  But I think we should state that.

Kevin-Andes, 2021-02-25:
In the TG discussion on 2021/02/16, Krste clarified that, if a higher privilege interrupt occurs, it will not resume WFI but start servicing this new high privilege interrupt instead.

Kevin-Andes, 2021-02-25:
Also, in the TG discussion on 2021/02/16, Bill Huffman had a nice proposal: use “0” for threshold to replace “-1” if we want the “no filtering mode,” which allows any interrupts to wake up WFI. This is cheaper (no cost at all) and also very intuitive since this usage is in agreement with the original definition.

billhuffman, 2021-02-25:
To clarify a bit, I believe using “0” would have exactly the obvious/expected behavior for the current privilege and higher privileges.  The difference if we do this would be that, when the threshold is set to “0”, WFI would be required to fall through on lower privileged interrupts as well.  When threshold is greater than zero, WFI would not be required to fall through on lower privileged interrupts (or on same privilege interrupts at or below the threshold).

The description is in terms of “would be required to fall through” (or not) because falling through is always allowed for any reason.  We’re only spec’ing when it’s required to fall through.

This keeps the interrupt selection as expected, but requires a small change in what actions are taken after the highest priority interrupt is selected.  It does mean that there’s no way to fall through on all current privilege interrupts but no lower privilege interrupts.

      Bill

EXTERNAL MAIL

Kevin-Andes, 2021-03-02:
Thank you for the detailed clarification. Indeed, using “0” does create a difference that WFI would be required to fall through on lower privileged interrupts (while the normal usage of qualifying preemption would not). Also, you are right that “there’s no way to fall through on all current privilege interrupts but no lower privilege interrupts” because the definition for “0” changes subtly now.

In order to keep it simple/backward-compatible and avoid potential confusion, it is probably better to keep the definition of “0” to be the same. To this end, adding a sign bit may be desirable and justified as we are not changing existing definition or behavior. 

By the way, the advantage of using a sign bit (i.e., negative threshold) is that it is somewhat easier to remember and program compared to an alternative of creating a separate mode bit elsewhere.

kasanovic, 2021-03-02:
This appears to be a more general issue than only for the CLIC.  Other interrupt controllers, including the original CLINT scheme, might want to have a way to change WFI behavior to ignore all lower-privilege interrupts when executed in a higher-privilege mode.  For example, we could add another mode bit in a status CSR to modify WFI behavior in this way for all local interrupt controllers.

If we address this in the CLIC spec, we might find that later we'll have to define the interaction of the CLIC setting with the other more-universal setting.  Suggest we raise this as a privilege group issue, and possibly worth fast tracking. OTOH this might be difficult to get agreement across the different local interrupt controller designs (CLINT, CLIC, AIA).

The WFI-mode bit when clear would preserve existing behavior.
The WFI-mode bit when set could have the form of 1) always ignore lower-mode interrupts, or 2) respect xie bits of lower-mode interrupts, so higher-mode can modify yie bits of modes y<x to get desired behavior.

Action is to raise this issue on priv committee.

kasanovic, 2021-03-05:
I had a discussion with Andrew and we came up with what we believe is a clean solution, which is to add a new WFMI instruction ("wait for mode's interrupts").  This instruction only has to wakeup for pending-and-enabled interrupts in the current mode, and is not required to wakeup for lower modes (pending-enabled higher-mode interrupts will just interrupt as usual).  Making a new instruction avoids having to deal with state (which could be separate bits for each privileged mode)  and we don't believe there is an important use case that wants to modulate the behavior of fixed binary code with WFI by changing a state setting.

The encoding is same as WFI except with bit 28 clear.  This also provides a true user-mode WFMI, whereas existing WFI is encoded as a supervisor mode instruction, with exception to allow use at user-mode.  The encoding also means a simple legal implementation can simply merge WFI/WMFI in hardware treating bit 28 as don't care (though losing benefit of selective wakeup).

@aswaterman Andrew will be bringing up privileged list for discussion as a general, hopefully fast-track, extension.

aswaterman, 2021-03-05:
Yeah, it will be a little while, but I'll get it out there.

billhuffman, 2021-03-05:
I think this allows a choice.  An implementation can do one of two things:
- Treat both instructions the same way.  Always wakeup on all interrupts of the current mode and on interrupts of all lower modes.
- Keep micro-state for which instruction was executed.  Do the same as above for WFI.  Wakeup only on above threshold current mode interrupts for WFMI.

Higher mode interrupts are always taken.

aswaterman, 2021-03-05:
Agreed.

dansmathers, 2021-04-13:
can issue #31 and #120 be closed with pull #144?

commit 7fa290c0 (2021-01-07): Specified the responses to Wait for Interrupt (WFI). Closed #31.