riscv/riscv-fast-interrupt #36: Interrupt level of vector table read operation

khannaudit, 2019-08-15:
In [Section 3.4](https://github.com/sifive/clic-spec/blob/master/clic.adoc#34-new-mtvec-csr-modes):

> For permissions-checking purposes, the memory access to retrieve the function pointer for vectoring is treated as a load with the privilege mode of the interrupt handler.

The specification does not specify what the "interrupt level" of this load should be. I think this is important to determine the interrupt level at which access exceptions and page faults for this vector table read should be serviced. [Section 5.3](https://github.com/sifive/clic-spec/blob/master/clic.adoc#53-synchronous-exception-handling):
> Horizontal synchronous exception traps, which stay within a privilege mode, are serviced with the same interrupt level as the instruction that raised the exception.

If we treat the load as the zeroth instruction in the interrupt handler, it makes sense to execute it at the level of the interrupt, but some clarification from the spec would be nice.

kasanovic, 2020-06-09:
Yes, same as interrupt handler.  Updating text to make clear.

closed 2020-06-09 by kasanovic

Kevin-Andes, 2020-06-16:
Namely:
For permissions-checking purposes, the memory access to retrieve the function pointer for
vectoring is treated as a load with the privilege mode and interrupt level of the interrupt handler.

commit ede21db9 (2020-06-09): Closes #36.