riscv/riscv-fast-interrupt #39: Proposal to remove trig field from clicintattr register

zavalt, 2019-10-28:
### Proposal to remove ‘trig’ field from clicintattr
The trig field consists of two definitions, interrupt polarity and trigger. The primary proposal is to remove both options as explained separately below, and consequently, removing the trig field altogether. 

As we come to a consensus, we may decide to keep a portion (or all) of the trig field definition. If this is the case then I would have a secondary proposal of defining the field as a WARL type in order to give a hardware-based option to not support it. Though this secondary proposal may seem attractive, I would argue that keeping the trig field will result in more complexity to the specification and cause all around more support effort and strongly encourage the primary proposal.

#a406f31674f6bab1164f3fb71874965ae28c1e1c
Section 3.7 CLIC Interrupt Attribute Register (clicintattr) has the following definition for the trig field:

> The 2-bit trig field specifies the trigger type and polarity for each interrupt input. Bit 1, trig[0], is defined as "edge-triggered" (0: level-triggered, 1: edge-triggered); while bit 2, trig[1], is defined as "negative-edge" (0: positive-edge, 1: negative-edge).

### Proposal to remove the interrupt polarity option. 
To ease both software and hardware complexity, the interrupts to the core shall always assert on the positive edge. Software developers will not need to query the system nor configure the core for the appropriate interrupt level. Hardware systems will always ensure interrupt polarity is active high to the core to comply to a consistent interrupt signal interface.

- Con: ?
- Pro: positive polarity is the normal case for most interrupt sources, reduces complexity, simplifies software programming model, less area, less verification, consistent implementation.

### Proposal to remove level-triggered interrupt option
To ease both software and hardware complexity, the interrupts to the core shall always be edge-triggered. The software developer will not need to query the system nor configure the core for the appropriate interrupt trigger. Interrupts asserted as a level from a module can be handled with edge-triggered configuration. If the interrupt source is still set at the return from an ISR then it will simply be taken again. This allows for a consistent hardware and software model.

Con: ?

Pro: Removing ‘level-triggered’ implementation simplifies hardware to one implementation including the clicintip behavior as defined in section 3.5 (CLIC Interrupt Pending). Software for core will be consistent for all ISR.

dansmathers, 2020-05-12:
Hi, the spec now says:
Note | Some implementations may want to save these bits so only certain trigger types are supported. In this case, these bits become hard-wired to fixed values (WARL).

can we close this issue?

kasanovic, 2020-05-12:
TG agrees with WARL definition.

closed 2020-05-12 by kasanovic