riscv/riscv-fast-interrupt #43: Is this specification still live?

JamesKenneyImperas, 2020-03-30:
Is this specification still live?

I have several questions about what is documented here, but I don't want to waste time posting if issues are not monitored.

Thanks.

jim-wilson, 2020-03-30:
The task group is still meeting twice a week.  There is a meeting tomorrow (Tuesday) morning.

JamesKenneyImperas, 2020-03-31:
Thanks, Jim, that is good news. I will go ahead and post my questions.

It seems odd that since the task group is still meeting so frequently there has been no specification change since December 8th and the open issues have not been touched for several months. Perhaps this could be raised as an issue to address at this meeting.

James.

closed 2020-03-31 by JamesKenneyImperas

jim-wilson, 2020-03-31:
You might try joining the fast interrupt task group, and then you can send questions to the mailing list.  That is more active than the github repo at the moment.