riscv/riscv-fast-interrupt #44: User mode interrupts and the N extension

JamesKenneyImperas, 2020-03-31:
Section 3.2.1 (Specifying interrupt privilege mode) which discusses interrupts for M, S and U levels appears to assume that User mode interrupts are always supported. I would have thought that the most typical system would support M and S level interrupts but not U level (N extension absent). Please can you clarify this section to describe this implementation case. For example, if the N extension is absent, is the `nmbits` field still two bits wide, and if so what happens if it is set to `00`?

Kevin-Andes, 2020-04-09:
No, we do not assume that User mode is always supported, and you are correct that most typical systems only support M and S level interrupts.

>For example, if the N extension is absent, is the nmbits field still two bits wide, 
>and if so what happens if it is set to 00?

Quick answer: If nmbits is set to 00, it means every interrupt is an M-mode interrupt (and there is no programmable registers exist in clicintattr[i].mode).

The nmbits is always 2-bit wide and its value indicates how many bits of physical registers (flip-flops) exist in clicintattr[i].mode. Basically, we try to save every bit of register (per interrupt) if possible.

In addition, although the nmbits is always 2-bit wide, it does not always contain 2 bits of physical registers (flip-flops). For example, if only M-mode is supported in the system, the nmbits can always be hardwired to 0 (since there is only 1 mode) so it does not contain any physical registers.  In this case, similarly, clicintattr[i].mode can also be hardwired to 0 so no physical register is present there either.

kasanovic, 2020-05-12:
Spec needs clarification.

Kevin-Andes, 2020-05-26:
Clarified in commit e29a07c481ae894fbc68d480705c926a19745d40.

closed 2020-05-26 by Kevin-Andes