riscv/riscv-fast-interrupt #45: xtvec behavior when base and original mode both supported

JamesKenneyImperas, 2020-03-31:
Section 4.4 (New xtvec CSR mode for CLIC) implies that, when both original and CLIC mode are supported, it is possible to use original interrupt mode for one exception level and CLIC mode for another. For example, `mtvec` could be configured for CLIC mode and `stvec` for original mode. What happens if this is done?

As a contrived example, suppose that `mtvec` is configured to use CLIC mode, and the CLIC routes the Supervisor timer interrupt to Machine mode, but that `stvec` is configured to use original mode to route the same interrupt to Supervisor mode. Which interrupt controller wins?

MoonbaseOtago, 2020-05-03:
Here's a more detailed (and related) note I had added to a closed bug that maybe explains this another way:

One issue that probably needs to be clarified is what happens when a mode is running in CLIC mode and a lesser mode is not (for example if M-mode is using the CLIC but S-mode isn't) then the spec says that xdeleg of the mode in CLIC mode (mdeleg in the example above) is specified to not have any effect ... this means that interrupts can't be passed (to S-mode in this case).

I sort of see two possible solutions to this:

either say it's undefined and that if this situation arises it's not required to work, and suggest that if a mode uses the CLIC all lesser modes should too, or
say that xdeleg has no effect on mode-x, but does have the effect of delegating to the next lesser mode (for example mdeleg would not necessarily disable any M-mode interrupts, but it would cause them to be passed to S-mode)

kasanovic, 2021-03-16:
For first rev 1.0, we can require that all privilege modes can only run in same mode, i.e., only one of CLIC or CLINT is visible in lower privilege modes depending on M-mode setting.  This should be added to spec.

There is a covert channel concern if additional state (i.e., sstatus.sie) is visible to lower privilege mode but not context swapped by higher privilege modes, so higher privilege mode would have to be aware that it allowed lower privilege modes to switch interrupt mode.  There is a more general discussion about delegation of feature mode bits happening in hypervisor group.

kasanovic, 2021-04-13:
Closed by pull #128

kasanovic, 2021-04-13:
The text should state that only M-mode can select CLIC versus CLINT mode.  Lower privilege modes will not have the ability to change between CLIC and CLINT modes.

kasanovic, 2021-08-17:
In current spec, if both CLIC and CLINT are supported, then bits[5:2] must be zero for both (i.e., CLINT *tvec must also be 64-byte aligned).  Other settings of bits[5:2] are reserved.  Bit 1 of *tvec is controlled by bit 1 of mtvec (i.e., lower privilege modes see this bit as read-only.  Bit 0 and bits[XLEN-1:6] of lower-privilege mode's *tvec are unaffected by changes to corresponding bits of mtvec.  These constraints might change if there are future additions to the CLIC or other new interrupt controller specs.

dansmathers, 2021-12-21:
in fast-interrupt TG meeting 12/21/2021, I thought Aug 17 comment was missing from pull #173.  It is included in the Note section above the xtvec[5:0] table.  I think no additional updates to pull #173 are needed.

kasanovic, 2022-01-04:
Closed by https://github.com/riscv/riscv-fast-interrupt/commit/7a45ea71da9b2bb6459038f365266b4af5ef2d73

closed 2022-01-04 by kasanovic