riscv/riscv-fast-interrupt #46: Recovery from table load fault

JamesKenneyImperas, 2020-03-31:
Section 4.4 (New xtvec CSR mode for CLIC) says:

> For permissions-checking purposes, the memory access to retrieve the function pointer for vectoring is treated as a load with the privilege mode of the interrupt handler. If there is an access exception on the table load, `xepc` holds the faulting address. If this was a page fault, the table load can be resumed by returning with `xepc` pointing to the table entry and the trap handler mode bit set.

Can you please give more information about how such recovery is possible? For example, suppose that an interrupt occurs at address `PC1`, and that while attempting to handle that interrupt there is a table load fault at address `TB1`. By the above description `xepc` is set to address `TB1`. Where is the original address `PC1` stored? How is return to this address handled?

kasanovic, 2020-04-14:
Should be handled by NMI if the fault was a machine error.  Otherwise, can trap into debug if run out of nested exception levels.

kasanovic, 2020-04-14:
Needs more clarification in spec.

billhuffman, 2020-05-14:
Sounds as if an access exception on a horizontal hardware vectored interrupt is fatal as xepc is lost.  That would match what happens if there's an access exception when saving to the stack while xepc is still live.  Is that the idea?

billhuffman, 2020-05-19:
I might expect that a page fault on the table access might set xinhv in xcause and set xtval to the faulting address while leaving other state the same.  For a horizontal interrupt, xepc contains the place to continue with the interrupted process and overwriting it seems fatal.

dansmathers, 2021-01-06:
Should we also add some clarification on this statement: 
For permissions-checking purposes, the memory access to retrieve the function pointer for vectoring is treated as a load with the privilege mode and interrupt level of the interrupt handler
I'd like to confirm that it is the privilege level of the load, i.e., we consider MPRV and SUM bits on that access.

kasanovic, 2021-04-13:
Table fault handling is covered in #131, so closing this issue.
Created new issue #142 to discuss MPRV and SUM.

closed 2021-04-13 by kasanovic