riscv/riscv-fast-interrupt #47: CLIC reset behavior

JamesKenneyImperas, 2020-03-31:
Except for one reference to the cliccfg register, there is no mention of reset behaviour in the specification. Is all other state unknown at reset?

kasanovic, 2020-04-14:
Needs clarification in spec.
General model in RISc-V has been mandatory reset state is minimized, platforms or company policy might add additional reset requirements.

kasanovic, 2021-03-16:
The general privileged architecture states that mstatus.mie is reset to zero, so coming out of reset interrupts will not be enabled.

For an S-mode only execution environment, the EEI should specify that status.sie is also reset on entry (it is then responsibility of EEI implementation to ensure that is true before beginning execution in S-mode).  Similarly for other lower-mode execution environments.

kasanovic, 2021-03-16:
The spec could be clarified to have a separate reset section covering these issues.

kasanovic, 2021-04-13:
Closed by pull #137

closed 2021-04-13 by kasanovic