riscv/riscv-fast-interrupt #48: Logic/state diagram for clicintip signals required

JamesKenneyImperas, 2020-03-31:
Section 3.4 (CLIC Interrupt Pending) attempts to describe the logic for `clicintip` signals, but I think this needs further clarification, either by a state diagram or a schematic. In particular, it says:

> `clicintip[i] `is a read-write register. Software-based (direct) writes to these pending bits have priority over hardware-based writes (triggers).

What does this mean if the input is configured as level-sensitive?

What state is latched and what is combinatorial? For example, what is the effect of reconfiguring a signal to be positive level-triggered instead of negative level-triggered while it is pending?

I think this section needs to fully describe the value of `clicintp` for every possible transition of the external input signal, corresponding `clicintie` and `clicintarttr` registers, and describe the effect of a software write to `clicintp` in every state.

anthony-coulter, 2020-05-22:
=== The CLIC triggering mechanism ===

The CLIC triggering mechanism for interrupt #i has the following components:

* The input signal from the interrupt source.
* Three configuration bits `trigger`, `polarity`, and `enable`
* An invisible storage bit `latch`
* The memory-accessible bit `pending`
* An input signal `autoclear` from the hart

The triggering mechanism thus latches four bits of state: `trigger`, `polarity`, `enable`, and `latch`.

=== Memory map of these bits ===

Four bits in the triggering mechanism are accessible via memory accesses:

* `trigger` is currently the second-least-significant bit of `clicintattr[i]`.
* `polarity` is currently the third-least-significant bit of `clicintattr[i]`.
* `enable` will probably be the least-significant bit of `clicintie[i]`.
* `pending` will probably be the least-significant bit of `clicintip[i]`.
* `latch` is not directly accessible in the memory map.

The value of the trigger bit is expressed with a symbolic constant: `LEVEL = 0` and `EDGE = 1`.

=== State of the bits on a reset ===

The state of the trigger, polarity, enable, and latch bits on a reset is unspecified. (The privileged spec already mandates that `mstatus.mie = 0` on reset, and that is enough to block all interrupts before they are configured by software.)

=== Reading these bits ===

Loads from the memory addresses corresponding to `trigger`, `polarity`, and `enable` read the saved bits as expected.

Loads from the memory address corresponding to `pending` returns the value:

        `pending := latch | (trigger == LEVEL & (source_signal ^ polarity)`

where `source_signal` is the signal from the interrupt handler.

When the CLIC determines whether an interrupt is pending for the purposes of taking an interrupt, it reads the `pending` bit the same way a memory access would.

=== Writing these bits ===

Stores to the memory addresses corresponding to `trigger`, `polarity`, and `enable` modify the stored values as expected.

A store to the `pending` bit changes the state of the `latch` bit.

Note that some of these bits may be hardwired, in which case storing to them has no effect.

=== Hardwiring considerations ===

The implementation MAY hardwire `trigger` to either `LEVEL` or `EDGE`.

The implementation MAY hardwire `polarity` to zero. It SHOULD NOT hardwire the `polarity` to one. (Rationale: Software can determine whether the polarity bit is reconfigurable by writing a 1 to it and seeing what happens. If the polarity bit is not configurable, there's no need for hardware to tell us whether the underlying physical signal is active-high or active-low, so we mandate that the hardwired value be zero.)

The implementation MAY hardwire `latch` to zero, but only if it also hardwires `trigger = LEVEL`. (Note that doing so makes it impossible for software to manually write to the pending bit.)

The implementation MUST NOT hardwire `enable` to zero. It MAY hardwire `enable` to one. (Hardwiring the enable bit to 1 creates a non-maskable interrupt. If we want to forbid that, then we should disallow hardwiring the enable bit to any value.)  

=== The autoclear signal ===

The `autoclear` signal is delivered to the interrupt triggering mechanism for interrupt #i under either of the following circumstances:

* `clicint[i].shv = 0` and interrupt #i is delivered to the hart.
* `clicint[i].shv = 1`, the hart writes to the `Xnxti` CSR, and the trap address returned corresponds to interrupt #i.

For ordering purposes: the hart writing to the `Xnxti` CSR is a regular instruction execution, and the hart accepting an interrupt must take place between some pair of consecutive instructions because interrupts are precise. Thus a hart cannot set the `pending` bit (with a store to `clicintip[i]`) concurrently with the delivery of the `autoclear` signal---one must logically precede the other.

=== Asynchronous changes to the latch bit ===

The `latch` bit can change under the following circumstances:
[... 30 lines cut ...]
1. A pure software interrupt:

Hardwire the input signal to one, `polarity` to zero, and `trigger` to whichever of `LEVEL`/`EDGE` encodes as zero. This has the effect of making the `pending` bit an alias of latch bit. Software interrupts are thus triggered only when a hart writes to the `pending` bit.

2. A cheap level-triggered interrupt:

Hardwire `trigger = LEVEL` and `latch = 0`. Reading the `pending` bit will return `xor(signal, polarity)`. Writing to `pending` has no effect.

3. Configuring an interrupt with a single store-word

A single store-word instruction can set `pending = 0`, `enable = 1`, and `trigger` and `polarity` to anything. This will clear the latch, enable the interrupt, and reconfigure trigger and polarity in such a way that no spurious interrupt is generated (because a write to the polarity bit MUST NOT be interpreted by hardware as an edge in the input line).

Note that in general, software is responsible for clearing the latch bit when it changes the trigger type.

=== Regarding the latch bit for level-triggered interrupts ===

This proposal allows software to generate a fake level-triggered interrupt by writing to the pending bit, which sets the latch and causes subsequent reads of the pending bit to return true. This looks crazy but my rationale is as follows:

1. Implementations are permitted to hardwire the latch bit to zero if they have also hardwired the trigger bit to LEVEL, so you can leave out most of the circuitry if you know you'll never use it.

[... cut ...] e to handle fake software-triggered interrupts it will notice that the pending bit is no longer set and assume that the input signal has dropped back to its original level.

4. As stated earlier, software can easily clear the latch whenever it reconfigures the trigger type in a single store-word instruction, so the burden of managing the latch during reconfigurations is minimal. In particular, clearing the latch on a reconfiguration in software requires less complexity than doing it in hardware.

JamesKenneyImperas, 2020-05-26:
This level of detail is excellent and exactly what is needed in the specification.

anthony-coulter, 2020-05-26:
There seems to be some resistance on the mailing list to the idea of keeping the latch around for level-triggered interrupts. To summarize my thinking about it (repeating a few points from the end of my proposal):
* The latch is optional for level-triggered-only interrupts; the proposal doesn't mandate that people add a latch to an implementation that doesn't need one.
* Edge-triggered interrupts and software interrupts both require a latch, so for configurable interrupts we need to specify the latch's behavior in level-triggered mode.
* This proposal gives a uniform description of how all interrupt types (level-triggered, edge-triggered, and software-written) can be expected to behave in the presence of a latch, while still allowing pure level-triggered interrupts to operate without a latch.

A counterproposal on the mailing list is to leave the latch behavior unspecified on reconfiguration. To fill in some details:
* The latch would be ignored in level-triggered mode; that is,
    pending := (trigger == EDGE & latch) | (trigger == LEVEL & (source_signal ^ polarity)
* The state of the latch after interrupts are reconfigured to `EDGE` is unspecified. (And the state of the latch while `trigger == LEVEL` is irrelevant.)
* Changing `polarity` while `trigger == EDGE` may or may not trigger a spurious interrupt. (This was not stated in the counterproposal, but I think it fits the spirit thereof.)
* To change the interrupt configuration deterministically, software should disable interrupts, then update `trigger` and/or `polarity`, and then then reenable/reinitialize interrupts.

I would push back on the third point: software should be able to reconfigure interrupts in a single instruction, so I would contend that a single store-word instruction to set `pending`/`enable`/`trigger`/`polarity` all at once ought to initialize the interrupt into a deterministic state.

silabs-PaulZ, 2020-07-10:
For the purpose of this discussion, I will use the term ‘sticky’ in lieu of ‘latch’.

In order to properly provide an edge-based interrupt, the CLIC will probably need to add three additional flops (or latches) per interrupt: 1) edge mode via clicintaatr.trig[0], 2)edge detection, 3)sticky capture. This is a little expensive for our case, a low power, low gate count embedded processor. Here’s an example implementation for discussion (not guaranteed, does not include all features such as SW setting/clearing pending )
[image]
With the current specification, the CLIC will be able to capture the three types of interrupts: level, edge, and pulse (the latter using the edge-based setting).

From an embedded point of view, we require pulse and level interrupts, but not the edge-based.  Enabling pulse and level will result in the 'expensive' edge-based mechanism.

In order to reduce the overall number of flops and provide a simple interrupt protocol, I propose to always capture the input interrupt into a sticky pending bit.
The proposed behavior is as follows: 
  - The interrupt will be captured into a sticky pending status when associated ISR is not active. 
  - Interrupt pending is automatically cleared by HW while the associated ISR is active.

Here’s an example implementation for discussion only.
[image]

Pros:
  - Removes the need for clicintattr.trig[0] edge/level mode flop
  - Removes the edge detection circuitry
  - Only one sticky capture needed to support pulse and level
  - The software does not need to clear pending (same as CLIC)
  - The software can always set pending directly (not shown above)
Cons:
  - Removes true edge-based interrupt detection

If this proposal is not accepted, we will need to define the behavior of the edge-based interrupt in more detail:

What is the expected behavior of the sticky/pending bit when an edge-based interrupt level is maintained high throughout the ISR?
[image]

What is the expected behavior of the sticky/pending bit when an edge-based interrupt occurs during the ISR (but after it was originally auto cleared). Is it expected to return to the ISR after it exits?
[image]

Should we instead clear the pending status at the end of the ISR?

Can people provide examples where they need a true edge-based vs. pulse?

kasanovic, 2021-08-03:
These issues were addressed in later edits to the spec.  However, Section 4.4 should be edited to be clearer about when hardware clears the IP bits as seen by software. Also, should refer to other sections that describe how edge-triggered interrupts are cleared.

Also for software clearing of the IP bit,  only writes to the xnxti can clear the bit, not read-only accesses.

Waveforms are too implementation-specific to be used in the spec.  Some cores will be asynchronous....

commit 876d8630 (2021-11-09): Indicate when edge-triggered interrupts are cleared.
Closes #48
closed 2021-11-09 by kasanovic