riscv/riscv-fast-interrupt #49: Clarification of memory map required

JamesKenneyImperas, 2020-03-31:
I think section 3 (CLIC Memory-mapped Registers) requires clarification. It would be helpful to add diagrams of the entire address map for M,U and M,S,U cases so that it is clear how these regions abut. A diagram is also needed for a multi-hart system. This information is implied in words, but a diagram would be better.

In a multi-hart system, how does a hart programmatically determine the address of its S and U regions? This requires knowledge of the number of harts in the system.

JamesKenneyImperas, 2020-03-31:
Additionally, could you please clarify whether the behavior of memory-mapped accesses is dependent on the hart execution mode or not? For example, does an access by a hart executing in Machine mode to a Supervisor mode memory-mapped register behave exactly in the same way as a Supervisor mode access? As a second example, suppose that PMP settings are misconfigured to allow User mode access to the Machine mode control block. Would such an access behave in the same way as a Machine mode access to that block?

I ask this because in some other similar systems (e.g. ARM memory-mapped interrupt controllers) both the privilege mode of the block _and_ executing processor are taken into account. It looks like this is not the case here, but I wanted to check.

kasanovic, 2021-03-16:
This is generally a platform issue.   The spec itself focuses on the structure within each aperture not the location of the apertures in the global physical memory space.  Each platform needs to define a discovery mechanism to determine the memory map locations.

We should remove or downgrade to commentary the recommendation on the layout of multiple harts and multiple privilege modes, as different platforms will want to handle these differently.

kasanovic, 2021-06-22:
We added text back in #157 to explain the different privilege-mode memory mode apertures.  Please reraise issue if this is still not clear.

JamesKenneyImperas, 2021-07-15:
I've only just got round to looking at this (sorry) and do have another question about the memory map.

In older versions of the specification, pages for a multiple-hart system were laid out so that there was an initial control page (containing `cliccfg` etc), then four M-mode pages for each hart, then four S-mode pages for each hart, and so on:

```
    control page 
    M pages x 4 (hart0)
    M pages x 4 (hart1)
    ...
    M pages x 4 (hart N)
    S pages x 4 (hart0)
    S pages x 4 (hart1)
    ...
    S pages x 4 (hart N)
```

This explicit layout has now been removed. Can I confirm that whereas previously there was a single control page for all harts in a system, now there will be a control page _per hart_, with each hart having a different `mclicbase`, like this:

    mclicbase (hart 0):      mclicbase (hart 1):     ...       mclicbase (hart N): 
    control page (hart0)     control page (hart1)    ...       control page (hartN)
    M pages x 4 (hart0)      M pages x 4 (hart1)     ...       M pages x 4 (hartN)

(S page locations will be defined in platform-specific manner.)

Thanks.

dansmathers, 2021-07-15:
reopening so we remember to respond to question in TG meeting

JamesKenneyImperas, 2021-07-20:
An additional note here for consideration at  the meeting - if the `cliccfg` register is replicated per-hart, then presumably it is now possible to configure `nmbits`, `nlbits` and `nvbits` differently for each hart as well, whereas previously when there was a single global configuration that applied to all harts. I'm not saying this is a problem - I just want to make sure that any issues that might arise because of this have been thought about.

kasanovic, 2021-07-20:
The CLIC spec will define the memory layout for each mode for a single hart, and not the memory map across modes or across harts.  The platform parameters or discovery mechanism will allow verification platforms or software to determine the memory map for multiple modes and multiple harts.

dansmathers, 2021-12-15:
created pull #183.  nmbits is only controlled by mcliccfg, nvbits is Read-only in mcliccfg/scliccfg/ucliccfg.  Only nlbits can be configured differently between priv modes.

kasanovic, 2023-01-31:
Closing this issue, and replacing with remaining concern about having separate cliccfg per privilege mode in issue #293.

closed 2023-01-31 by kasanovic