riscv/riscv-fast-interrupt #51: Add xscratchcsw/xscratchcswl  read and write pseudocode

JamesKenneyImperas, 2020-04-07:
The specification includes pseudocode for accesses to `xscratchscw` and `xscratchscwl` using read-modify-write accesses. For clarity, please also add pseudocode for read accesses and write accesses.

Kevin-Andes, 2020-06-23:
We will discuss this further in our discussion group.

kasanovic, 2020-07-07:
Kevin's email:
Hi,
 
In our recent group meeting, we discussed about Jame’s github issue #51 but decided to get more feedback here because the answer seems somewhat unintuitive.
 
This issue asks about the behavior (pseudocode) of “pure reading” and “pure writing” xscratchcsw/xscratchcswl CSRs. These 2 CSRs are designed to speed up the initialization of the stack pointer by conditionally swapping the stack pointer with xscratch. Therefore, their intended/typical usage is read-modify-write (i.e., swap):  csrrw sp, mscratchcsw, sp
 
The pseudo code for “csrrw rd, mscratchcsw, rs1” (conditional swap) is the following:
 
// Pseudocode operation.
if (mcause.mpp != M-mode) then {
t = rs1; rd = mscratch; mscratch = t;
} else {
rd = rs1; // mscratch unchanged.
}
 
If we follow the same spirit of “conditional swap,” the write access seems to also become a “conditional write” with the following pseudocode:
 
csrrw x0, mscratchcsw, rs1
 
// Pseudocode operation.
if (mcause.mpp != M-mode) then {
mscratch = rs1;
} else {
// mscratch unchanged.
}
 
 
Similarly, read access seems to become a conditional read:
 
csrrs rd, mscratchcsw, x0
 
// Pseudocode operation.
if (mcause.mpp!=M-mode) then {
rd = mscratch;
} else {
// rd unchanged
}
 
Are these the desired behavior/definition for writing and reading xscratchcsw/xscratchcswl? Although pure reading or pure writing these CSRs may not be very useful, we need to define them clearly to get consistent results/implementation. In practice, the only useful access for these 2 CSRs should be read-modify-write (swap).
 
Thanks.
 
        Regards,
        Kevin

kasanovic, 2020-07-07:
In TG meeting, we decided to write out all the behaviors of the CSR variants, but could just possibly simply declare the non CSRRW variants as "reserved".

Kevin-Andes, 2020-08-18:
If we follow the same spirit of “conditional swap”, read access should become a conditional read:

csrrs rd, mscratchcsw, x0

// Pseudocode operation.
if (mcause.mpp!=M-mode) then {
    rd = mscratch;
} else {
    // rd unchanged
}

However, a senior member indicated that keeping the value unchanged (in false clause) will require an extra GPR read-port in an out-of-order CPU. This is quite expensive so another member suggested to always write the destination with x0 (i.e., rs1/source GPR). Nevertheless, for set/clear CSR instructions (csrrs/csrrc), directly copying the source GPR to the destination GPR will require an complete new datapath/MUX which did not exist before. This extra datapath and circuit may slow down the CPU max frequency.

In practice, only the read-modify-write (swap/CSRRW) operation is useful for xscratchcsw/xscratchcswl. In contrast, the “pure read” and “pure write” operations are meaningless and not useful at all. Therefore, we could just possibly simply declare the non CSRRW variants as "reserved". In the meantime, we will still try to write out all the behaviors of the CSR variants for further discussion.

dansmathers, 2021-03-04:
added pull request #119.  please review.  Thanks!

Kevin-Andes, 2021-03-11:
Updated the spec with commit 646310a5e4ae055964b4680f12c1c04a7cc0dd56.

closed 2021-03-11 by Kevin-Andes