riscv/riscv-fast-interrupt #55: cliccfg is listed as RW but cliccfg.nvbits is read-only

dansmathers, 2020-04-15:
Posting as issue to track.

I have another question.   This is about cliccfg description:

This says cliccfg is RW, 

0x0000       1B          RW        cliccfg
but nvbits is read-only. 

"The single-bit read-only nvbits field in cliccfg "

You are correct that defining nvbits as read-only is inconsistent with the RW attribute of cliccfg. Nevertheless, this is related to our discussion in the last meeting about whether to support 1 mode or 3 modes (or even 2 modes). Originally nvbits was read-write but I simplified it to read-only. We definitely need to revise this in the near future after we have determined how many modes are mandatory and supported.

kasanovic, 2020-04-28:
Need to clarify that not all bits are reset to zero, only the writable bits.

closed 2020-05-26 by Kevin-Andes

Kevin-Andes, 2020-05-26:
Fixed in commit dd0fdd9893f77cbbe58c9dbf59d9a7eb33c1034c