riscv/riscv-fast-interrupt #58: missing xintthresh from table in section 6.1 summarizing overall behavior?

dansmathers, 2020-04-20:
In section 6.1. General interrupt overview:
The overall behavior is summarized in the following table: the Current p/ie/il fields represent the current privilege mode P (not software visible), interrupt enable in xstatus ie and interrupt level L in xintstatus; 
Should it instead say:
interrupt enable in xstatus ie and interrupt level L max(xintstatus.mil, xintthresh.th );

dansmathers, 2020-04-24:
I also noticed it looks like 'ie' in the table should represent interrupt enable in xstatus.ie & clicintie[i].  only xstatus.ie is mentioned in the spec in this section.

kasanovic, 2020-04-28:
Text and table needs clarification.

dansmathers, 2021-03-04:
added pull request #118.  please review.  Thanks!

Kevin-Andes, 2021-03-11:
Updated in spec checkin c26f4df2d57bd42f7f346d864725cfa640552b6f.

closed 2021-03-11 by Kevin-Andes