riscv/riscv-fast-interrupt #61: 20190610 change #2: Added optional support for Non-Maskable Interrupt (NMI)

dansmathers, 2020-04-24:
Draft 20190610 – separating multiple changes as individual issues to enable change tracking/approval from draft-20190610 doc upload. 

Added optional support for Non-Maskable Interrupt (NMI) with encoding of mcause.exccode defined as 0xFFF and mcause.Interrupt=0. 

Comments from Krste 4/1/2020:
This CLIC NMI is not aligned with rest of spec - please disregard.

There is a separate more complete NMI proposal that is intended to be
used with all RISC-V systems, including those without CLIC, to be
debated in a priv committee TG.  This adds resumable NMIs (the current
RISC-V spec has non-resumable NMIs) with separate cause/epc.

kasanovic, 2020-04-28:
More standard NMI doc to be posted in this TG then up to privileged SC.

kasanovic, 2021-02-16:
Closed by #112

closed 2021-02-16 by kasanovic