riscv/riscv-fast-interrupt #75: 20200423 change #6: Rewrote and reordered some paragraphs to improve readability

dansmathers, 2020-04-24:
Draft 20200423 – separating multiple changes as individual issues to enable change tracking/approval from draft-20200423 doc upload.

20200423 change #6: Rewrote and reordered some paragraphs to improve readability

Overview section now discusses 4096 interupts and clicintip, clicintie, clicintattr, clicintctl immediately instead of waiting for clicintip/clicintie/clicintattr/clicintctl description sections.

Note: Krste mentioned in meeting 3/31 that we should have the spec say non-vectored design should be explained first.   The others should be added later with separate descriptions to avoid confusion and to make it easier to change the spec to make hardware vectoring optional if we decide later.  20200423 change 6 did not address this.

In draft 20200423, hw vectoring is mentioned in
1. Background and Motiviation - "The CLIC also supports a new Selective Hardware Vectoring feature that allow users to optimize each interrupt for either faster response or smaller code size."
3.2 CLIC Configuration (cliccfg) - nvbits description
3.2.4:  Specifying Support for Selective Interrupt Hardware Vectoring
3.4: CLIC Interrupt Pending (clicintip) - clearing mechanism for edge-triggered shv
3.6: CLIC Interrupt Attribute (clicintattr) - description of shv field
4.4: New xtvec CSR Mode for CLIC-  when mtvt is selected instead of mtvec
4.7: Changes to xcause CSRs - xinhv descriptions
4.8: Next Interrupt Handler Address and Interrupt-Enable CSRs (xnxti) - shv does not show up in xnxti
6.1. General Interrupt Overview - V representing the result of any hardware vectoring
7.2. Inline Interrupt Handlers and "Interrupt Attribute" for C - "With hardware vectoring, inline interrupt handlers can provide very rapid response for small tasks."
8.1. C-ABI Trampoline Code - multiple cases how xnxti behavior is affected by shv and non-shv interrupts.
9. Interrupt-Driven C-ABI Model - interaction between non-shv and shv

dansmathers, 2020-04-28:
renaming issue.  was: 20200423 change #6: Rewrote and reordered some paragraphs to improve readability.

based on discussion 4/28/2020:
List software vectoring first in spec (assumed to be mandatory in all cases)
Hardware vectoring (inc selective hardware vectoring) described afterwards in same spec

Not a decision what’s mandatory just a reorg of document

kasanovic, 2023-01-31:
PR #292 addresses this.

closed 2023-01-31 by kasanovic