riscv/riscv-fast-interrupt #76: stvec/utvec

MoonbaseOtago, 2020-04-27:
Quick question - does it make sense to have machine mode running in CLIC mode and S/U modes running in non-CLIC mode? (or any other arbitrary combination)

Or to put it another way does setting the lower 2 bits of mtvec override the lower 2 bits of stvec and utvec? or are they independent?

JamesKenneyImperas, 2020-04-28:
The specification as written implies that these are independent, but I also think this needs clarification if it is desired behaviour.

MoonbaseOtago, 2020-04-28:
Yes I agree, I think the important thing is that everyone implements the same thing, I'm not sure I see an advantage either way

dansmathers, 2020-04-28:
Hi, we discussed this in our last meeting and agreed we need clarification in the spec.  Can we combine this issue with: #45 xtvec behavior when base and original mode both supported for tracking?

kasanovic, 2020-04-28:
This issue is captured by #45 so closing this one.

closed 2020-04-28 by kasanovic

MoonbaseOtago, 2020-05-03:
One issue that probably needs to be clarified is what happens when a mode is running in CLIC mode and a lesser mode is not (for example if M-mode is using the CLIC but S-mode isn't) then the spec says that xdeleg of the mode in CLIC mode (mdeleg in the example above) is specified to not have any effect ... this means that interrupts can't be passed (to S-mode in this case).

I sort of see two possible solutions to this: 
1) either say it's undefined and that if this situation arises it's not required to work, and suggest that if a mode uses the CLIC all lesser modes should too, or
2) say that xdeleg has no effect on mode-x, but does have the effect of delegating to the next lesser mode (for example mdeleg would not necessarily disable any M-mode interrupts, but it would cause them to be passed to S-mode)

dansmathers, 2020-05-03:
Hi Paul,
Do you want to add this comment to issue:
https://github.com/riscv/riscv-fast-interrupt/issues/45
xtvec behavior when base and original mode both supported #45

This seems related.
Thanks,
Dan

MoonbaseOtago, 2020-05-03:
Sure - done - thanks