riscv/riscv-fast-interrupt #77: prev issue 57 - 1: additional detail on CLIC M/S/U memory mapped regions

dansmathers, 2020-04-28:
Separating previous issue #57 into multiple issues for better tracking
1. location
I want to reiterate the issue around knowing where a hart's CLIC M/S/U regions are located - I'd like to suggest that we probably need not just mclicbase but sclicbase/uclicbase (where appropriate) too - they're cheap because they're read-only

kasanovic, 2021-03-16:
I don't think we need these CSRs.  Most embedded systems will compile in the CLIC memory map as a constant instead of dynamically querying this value and using as a variable.  Others are working on more general discovery mechanism for debug tools that would not need to chew up CSR address space.  

The lower-mode base as seen by lower-mode code can be affected by address translation of lower modes.

dansmathers, 2021-04-13:
can this be closed with pull #129 ?

commit 935c046c (2021-07-06): Clarified text on memory map for different privilege modes.
Closes #156.
Closes #77.
Closes #79.
closed 2021-07-06 by kasanovic