riscv/riscv-fast-interrupt #80: prev issue 57 - 4: the version fields in clicinfo

dansmathers, 2020-04-28:
Separating previous issue #57 into multiple issues for better tracking
4. the version fields in clicinfo
We probably need some exposition here:
a) the "architecture version" presumably relates to a particular version of this spec - the spec should state something like "CLIC implementations that conform to this version of this spec shall have an architecture version of '1'" (or whatever number you decide
b) I'm unsure what the "implementation version" field is for - is that free for the implementer to use?

kasanovic, 2021-03-16:
clicinfo falls into the category of platform discovery info.  If there is agreement on discovery mechanism, this data could move there out of CSR space.  Most embedded platforms will compile these constants into code instead of querying the register, and it is primarily debug tools that want to ascertain these parameters.

Hopefully a separate discovery mechanism will have fewer constraints on encoding space for these fields.

kasanovic, 2021-07-06:
Discovery TG is asking for other proposals in flight to list the set of discoverable parameters for the extension.  Should probably add note to clicinfo and clicbase (and anywhere else a parameter is lurking) saying this will should be replaced with the general discovery mechanism.

kasanovic, 2021-10-12:
Should determine what are the basic parameters from which others can be determined in Chapter 6.  These are the ones that should get included in whatever discovery mechanism is eventually created for each platform.

kasanovic, 2021-11-09:
Closed by https://github.com/riscv/riscv-fast-interrupt/commit/d4688482b0cc256302faee65b1390842876c2e1b

closed 2021-11-09 by kasanovic