riscv/riscv-fast-interrupt #81: prev issue 57 - 5: programming CLICs in S and U modes

dansmathers, 2020-04-28:
Separating previous issue #57 into multiple issues for better tracking

5. programming CLICs in S and U modes
I don't think that a programmer can program a CLIC in supervisor and user modes without knowing the contents of cliccfg and clicinfo - I think that they should be available read-only in S and U CLIC spaces
Equally I'm not sure that poking through the interrupt space to find out which interrupts have been configured for your mode is all that safe, it might be appropriate to have a bit that's visible if an interrupt is available (say bit 3 of clinintattr)

kasanovic, 2021-03-16:
This falls into general category of delegating configurable features to lower privilege modes.  There is discussions in hypervisor group about a more general pattern for these kinds of settings.

For now, the S-mode execution environment interface can define how the interrupt controller is made visible to S-mode for example without requiring additional S-mode CSR access.

kasanovic, 2021-11-23:
yes, I think the other issues resolved this.

closed 2021-11-23 by kasanovic