riscv/riscv-fast-interrupt #82: xcause registers

MoonbaseOtago, 2020-05-06:
I've been writing interrupt test code .... and it dawns on me that adding new non-0 bits to the xstatus registers may break existing trap handler code (for exceptions, not for interrupts)

It might make sense to create new xistatus registers that are the same as xstatus but with the new fields

(note: I seem to be making a lot of issues here, please poke me if I'm being too annoying)

Paul Campbell

brucehoult, 2020-05-07:
Hi Paul, I believe that would make such code already broken -- you're supposed to ignore reserved fields.

p.s. good to see you here. Let me know if you get to Wellington, once that is permitted...

MoonbaseOtago, 2020-05-07:
(yeah hi!) I guess it mostly depends on how existing code is written, at worst it's one extra instruction in the system call path

MoonbaseOtago, 2020-05-08:
BTW I guess that means the current linux entry.S is broken :-)

aswaterman, 2020-05-08:
Linux entry.S is actually a red herring, because it's writing sstatus from a previous value stored to memory.  The `start_thread` code that initializes that value in memory is the potentially problematic thing.  Point being, there's no need to add extra code to the critical code path in Linux.

It shouldn't be problematic in practice anyway because new bits will be added to sstatus in such a way as to not break existing software.

MoonbaseOtago, 2020-05-09:
sstaus is not the issue ... it's scause in:

handle_exception:
  .....
   csrr s4, CSR_CAUSE
   .....
   bge s4, zero, 1f
  .....
1:
  ....
  li t0, EXC_SYSCALL
  beq s4, t0, handle_syscall

The problem is that because we're adding scause.mpp/mpie/mpil then s4 needs to be masked before anyone checks it's value (the bge is OK though) - as i said it needs an extra masking operation before use

aswaterman, 2020-05-09:
The Linux SBI can (and should) mandate no-CLIC mode.

dansmathers, 2020-05-12:
FYI, this is related to issue #30 which asks when an exception is taken in CLIC mode, does mcause look like non-clic or clic.   This issue #82 assumes it looks like clic, but it should be clarified in the spec.

MoonbaseOtago, 2020-05-14:
Note: if you can't see the saved mcause.mpil you can't save it (is it still there or is it set to 0?) - this is of particular interest if you take a trap (for example a page fault) while accessing a clic vectored interrupt

kasanovic, 2021-03-16:
For rev 1.0, we will mandate that all privilege modes see the same core-local interrupt controller (CLIC or CLINT), with only M-mode able to support configuring the interrupt controller.
See also #45.

Handling the different visible state in each privileged mode would be part of a potential future design supporting per-mode CL interrupt controllers.

kasanovic, 2022-01-18:
Moved to a separate issue #192  to be handled post-1.0.

closed 2022-01-18 by kasanovic