riscv/riscv-fast-interrupt #84: xtvt alignment

MoonbaseOtago, 2020-05-14:
(still working on writing test code for my implementation, almost done)

xtvt alignment is completely open (must by >= 2**6) but there's no upper limit .... but practically there is, we're limited to 4096 interrupts so 2**15 for RV64 and 2**14 for RV32.

Programmers who are writing code that create low level interrupt stuff are often flying with few resources, in particular heaps and things they can do random allocation on, frankly they're going to look at this and ask themselves "what '.align' directive can I use that will work everywhere? I'll let the linker worry about it"

I think the spec should require that xtvt should be speced to require at most 2**15 (32k) alignment (so it supports at least bits [SV-1:15] and at most [SV-1:6])

Kevin-Andes, 2020-06-23:
There is no requirement for maximum alignment. It is implementation dependent and will not be specified. It is like xtvec. Any bits above minimum alignment can be WARL. SW can determine the actual alignment by writing all 1's then read them back.

closed 2020-06-23 by Kevin-Andes

Kevin-Andes, 2020-06-23:
Programmers can define a parameter for '.align' in the header file. This parameter should be derivable from the configuration structure in the future.