riscv/riscv-fast-interrupt #85: initialising current interrupt level

MoonbaseOtago, 2020-05-16:
xintstatus is currently documented to be read-only  .... at some point in booting an OS the current interrupt levels will need to be set to 0 (so that interrupts can be serviced) as they are currently undefined at reset. The only current way to do this is to write xcause.pil and then execute an xRET instruction - it's particularly difficult for m-mode to do this for say u-mode

I think it would be useful to make xintstatus writeable - it makes OS setup easier (along with writing test cases)

Kevin-Andes, 2020-07-20:
Actually, the following statement is probably not true: "at some point in booting an OS the current interrupt levels will need to be set to 0 (so that interrupts can be serviced) as they are currently undefined at reset."

This is because, upon new interrupts, HW will automatically record the correct interrupt level information and thus does not depend on SW to initialize xintstatus. In other words, there is no need for SW to initialize xintstatus.

Kevin-Andes, 2020-07-21:
Please refer to issue #88.

closed 2020-07-21 by Kevin-Andes