riscv/riscv-fast-interrupt #86: xnxti and clearing pending for edge-triggered interrupts

billhuffman, 2020-05-19:
I believe we've discussed having an access to xnxti (that includes a write) clear the pending bit for edge-triggered interrupts.  But section 4.8 doesn't say anything about that.

dansmathers, 2020-05-26:
Section 3.4 discusses this:
In contrast, when non-SHV (common code) interrupt is selected, the hardware will not automatically clear the pending bit in edge-triggered mode. Instead, hardware will clear the corresponding pending bit only when software uses a csrrsi/csrrci xnxti instruction to select this interrupt and return its entry address. However, if the CSR instruction does not include write side effects (e.g., csrr t0, xnxti), then no state update on any CSR occurs and thus the interrupt pending bit is not cleared. This behavior allows software to optimize the selection and execution of interrupts using xnxti.

but I agree it makes sense to mention this in the xnxti description in section 4.8.

billhuffman, 2020-05-26:
It may make sense only to mention xnxti in section 3.4 and to include all the details like with and without state updates in the xnxti section.

kasanovic, 2021-03-16:
Closed with #123.

billhuffman, 2021-07-26:
Section 5.8 does now say that an edge-triggered interrupt will be automatically cleared.  But I don't think it's very clear yet.

1. It says it's cleared only when software uses an csrrsi/csrrci xnxti.   But I expect it's also cleared on a csrrw xnxti as well.  And on a csrrs/csrrc with source x0.  And not on a csrrsi/csrrci with immediate #0.
2. The pseudo-code says nothing about clearing the interrupt.

kasanovic, 2021-08-03:
The pseudocode in 5.8 is incorrect in that it shows side effects occurring even when uimm =0.
Pseudocode should also show edge-triggered interrupt bit being cleared down when interrupt is taken.

The section should be expanded to include effect of all types of CSR instructions (i.e., CSRRW, CSRRWI, CSRRS, CSRRC not just CSRRSI/CSRRCI).

Expectation is that behavior with other instructions is same as for CSRRSI/CSRRCI versions except for different source of operand values.

commit 173fc39f (2021-08-31): Update mnxti pseudo-code to handle side-effects correctly.
Closes #86.
Closes #165.
closed 2021-08-31 by kasanovic