riscv/riscv-fast-interrupt #87: Clarify how page faults are handled on vector table reads

kasanovic, 2020-06-09:
Not clear how to recover if EPC written with table entry location.

billhuffman, 2020-06-09:
One possibility is that we don't write anything except xcause[30] on a page fault (or other exception on the table lookup).  Xcause[30] says there's a fault on the table lookup.  From that, and the interrupt number in xcause, software can determine the location of the fault as well as how to restart.

Another possibility is that xtval contains the location of the fault (like it would for other faults).

Using EPC seems odd as well as seeming like recovery isn't possible.  If it does work, some non-normative text is needed to say why recovery is possible and why it makes sense to use EPC.

MoonbaseOtago, 2020-06-11:
In general this is a difficult problem - we're all building RISC cores, and this is a CISCish sort thing - as an example in my (highly speculative) core an interrupt forces a 'trap' uop into the pipeline - that uop has the future EPC - and for most interrupts acks the highest pending interrupt and then traps - but in the vectored case that uop when executed has to become 2 more uops - a load followed by a branch - the load might in turn trigger a page fault at the supervisor and/or hypervisor level - if that happens you want to leave the CPU in a state when returning from that page fault restarts the interrupt processing again 

Unlike most almost every other case in a RISCy core these have to involve hidden state across uops, the CPU can't ack the interrupt with the CLIC until the 'branch' is executed (otherwise the interrupt can't be restarted after the page fault is resolved), equally the interrupt state has to be frozen at the start of the process to make sure you're acking the same interrupt you fetched the vector for .... lots of potential evil error cases (hidden state is bad) 

I don't actually expect a supervisor interrupt vector fault to be able to be usefully handled with a supervisor page fault (how are you going to do disk IO to pull in a page containing the vectors if you can't take interrupts because you don't have vectors?). What's more useful (and important) is being able to handle user mode interrupts in supervisor mode or supervisor interrupts in hypervisor mode.

However there are some somewhat spurious cases that should be avoided, in particular systems software people should be cautioned to set the ACCESSED bit in every PTE traversed when fetching an interrupt vector .....

kasanovic, 2021-02-16:
A horizontal trap on vector table access (e.g., interrupt taken in M-mode with table not accessible to M-mode) is a fatal error, for which recovery is not expected (similar to avoiding exceptions in the early part of any ISR).  Vertical traps to a higher-privilege mode (e.g., from S-mode interrupt handler to HS-mode fault handler) write hsepc, preserving sepc, so the S-mode handler can be restarted after table is paged in for example.

This explanatory text needs to be added to spec.

Separately, there is a concern about why xepc is written and not mtval (#111).

kasanovic, 2021-04-13:
Closed by #131

closed 2021-04-13 by kasanovic