riscv/riscv-fast-interrupt #88: xintstatus CSR numbers

JamesKenneyImperas, 2020-06-22:
The new xintstatus CSRs are entirely read-only, but section 4 implies addresses  0x347, 0x147 and 0x047 (for mintstatus, sintstatus and uintstatus, respectively). Is this correct, or should the CSRs be in the read-only address range (i.e. addresses 0xF47, 0xD47 and 0xC47)?

Kevin-Andes, 2020-06-23:
These CSRs are temporarily allocated in the read-write address range because we may have more read-write fields in the future. We may change this before we ratify the spec.

Kevin-Andes, 2020-07-21:
Intentionally leaving this issue open just to remind us to review these CSR address mapping before ratification.

Kevin-Andes, 2020-07-21:
#85 mentioned the following statement, but it is probably not true: "at some point in booting an OS the current interrupt levels will need to be set to 0 (so that interrupts can be serviced) as they are currently undefined at reset."

This is because, upon new interrupts, HW will automatically record the correct interrupt level information and thus does not depend on SW to initialize xintstatus. In other words, there is no need for SW to initialize xintstatus.

dansmathers, 2022-11-08:
closed by #285

closed 2022-11-08 by dansmathers