riscv/riscv-fast-interrupt #89: Proposal: reorder lower 16 bits

silabs-PaulZ, 2020-06-30:
Regarding section: 13. CLIC Interrupt IDs

Being that software will need to handle interrupts differently in CLIC mode vs basic mode, why do we need to have the lower 16 CLIC IDs match the basic interrupt controller?

The existing definition of the lower 16 interrupts have the following downsides:
  - Each of the legacy timer (xTIP), software (xSIP), and external (xEIP) interrupts have essentially a redundant alias for each mode (user, supervisor, machine) that can functionally overlap based on clicintattr (section 2.2 CLIC Interaction with Other Local Interrupts)
  - The order is not intuitive as software interrupt should be a higher priority than the timer interrupt, but it is placed lower in the list. 
    - Note: the basic interrupt order does not match priority, while CLIC priority is based on order
    - To achieve a higher software priority, the CLIC spec adds another software alias, CSIP (essentially 4 aliases)
  - Vectored interrupts will have several holes in the bottom of the vector table that will waste memory. 

Proposal option 1 (preferred by author) is to simply remove the requirement for a predefined timer, software, and legacy external interrupt. Many embedded applications do not care for this predefined mapping. Each device would be responsible for providing a software interface header defining the location of each interrupt.

Proposal option 2 is to reorder the legacy interrupt locations. This would be a guideline as opposed to a requirement.
  - 0 : timer
  - 1 : software
  - 2 : external (legacy)
  - 3-4095: external+

Both proposals would allow for the removal of the additional aliasing of legacy interrupts, including removing CSIP,  and provides an optimized memory footprint for vectored interrupts.

Kevin-Andes, 2020-07-07:
We will discuss this issue further in the discussion group.

kasanovic, 2020-07-21:
Would like to have lowest priority being a software interrupt used for local context switching, but also desire a higher-priority software interrupt (above context switch timer) used for inter-core signaling.

kasanovic, 2020-07-21:
A fixed new ordering (proposal 2)  is a subset of the "anything goes with a configuration description" model (proposal 1).  For portable software, the fixed ordering is preferable.  Proposal 1 supports backwards compatibility.

A packed ordering saves vector table memory size.  Proposal 1 supports denser tables when not all interrupts are required, whereas proposal 2 mandates a few interrupts.

kasanovic, 2020-07-21:
The legacy external can be folded into the other external (perhaps by convention always at the lowest number?)

kasanovic, 2020-07-21:
Machine-mode should be able to see the lower-privilege mode interrupts with separate IDs, when not delegating the interrupts.
If there are no lower privilege modes, can pack as in proposal 2 above.

M-mode interrupt map (in M-only or M/U system without N extension)
0: M-mode software interrupt
1: M-mode timer interrupt
2+: external (including legacy)

External interrupts include inter-processor signaling interrupts.  Dedicated software interrupt generally expected to be reserved for local software scheduler.

With more privilege modes, need to have lower-privilege mode interrupts visible at lower fixed priorities than M-mode interrupts when not delegated (as in original CLINT design, but not necessarily at same interrupt IDs), e.g.,

M-mode interrupt map (in M/S/U system without N extension)
0:  S-mode software interrupt
1:  S-mode timer interrupt
2: M-mode software interrupt
3: M-mode timer interrupt
4+: external (including legacy)

S-mode interrupt map (in M/S/U system)

0: Software interrupt
1: timer interrupt

This would be the recommended layout, but not clear if this should be mandated.  Could be part of a profile for given set of RTOS to require this layout.

kasanovic, 2020-08-25:
This issue was raised in #59.
Might want to in addition specify a standard allocation for the PLIC external interrupts inputs, if they are present.

0: S-mode software interrupt
1: S-mode timer interrupt
2: S-mode external (PLIC) interrupt
3: M-mode software interrupt
4: M-mode timer interrupt
5: M-mode external (PLIC) interrupt
6: external

Not clear if this needs to be mandated, or just a recommendation.

kasanovic, 2020-10-20:
We should update the spec to indicate these orderings are recommendations as part of a profile, not mandatory in all incarnations of the CLIC.

kasanovic, 2021-02-17:
Closed by https://github.com/riscv/riscv-fast-interrupt/commit/0af35df5fdf3d6103500de98ad6e8352c8c94e82

closed 2021-02-17 by kasanovic