riscv/riscv-fast-interrupt #90: clicint* reserved and custom bits and software use of inactive entries

David-Horner, 2020-07-07:
clicint* byte fields contain a single authoritative bit.

However, future extensions may expand the bit field or provide additional info/control through use of further bits.

Just as with map table entries have reserved bit settings (e.g. if the entry is inactive, other fields are available to software), so I suggest a similar approach.
that we both reserve some bits for software use, and for inactive elements software may use other bits if they are provided in hardware.

I propose we standardize on the structure and nature of reserved bits and custom bits in the clicint* fields (and any others similarly defined in the clic proposal).
Specifically, 
- reconsider/weighin on  bit 7 as the primary control/status bit.
- allow any other R/W hardware bits for software use if control bit(s) denote inactive.  
- reserve  a set of otherwise unused bits.
- designate as custom a set of otherwise unused bits.

David-Horner, 2020-07-07:
this covers some of the issue #41

kasanovic, 2020-07-07:
Conversation in TG. 

Ip is not frequently read by software - the main function is to cause a hardware interrupt.  It will be more often written, either by hardware or software.

Because ip byte is written by hardware and software, we want to avoid packing bits unrelated to signaling an interrupt in that byte.  any additional bits should be related to the pended interrupt.  The text should be updated to indicate that ip=zero means no interrupt pending, and ip!=0 (not just 1) indicates an interrupt is pending.

A minor point, but writing a 1 is easier in RISC-V software than writing 0x80, as 16b instruction can create a 1 but need 32b instruction to create 0x80.

kasanovic, 2021-02-17:
Closed by https://github.com/riscv/riscv-fast-interrupt/commit/664f438c1990289c3c5daff1483cd1ae4a59fff5

closed 2021-02-17 by kasanovic