riscv/riscv-fast-interrupt #92: Hypervisor compatibility

JamesKenneyImperas, 2020-07-30:
Is this specification known to be compatible with the developing Hypervisor extension? I would have thought that support might be needed to allow the CLIC to be easily virtualized, but I can't see anything about that here.

Thanks.

JamesKenneyImperas, 2020-07-30:
It seems that interrupt 12 is used here for the CSIP interrupt but also in the Hypervisor specification for supervisor level guest external interrupts.

kasanovic, 2020-08-25:
There is discussion among hypervisor authors on using a subset of CLIC functionality as basis of virtualized interrupt capabilities., but I have not seen anything written down yet.  The goal is not really to virtualize the CLIC per se, but to allow virtualization of device interrupts.

kasanovic, 2021-11-23:
The AIA spec is being developed to be the standard support for interrupts for standard hypervisors.   Possible hypervisor directions for fast interrupts for embedded include: 1) a minimal CLIC extension that adds HS mode as another privilege layer in the stack, so the hypervisor could optionally receive interrupts and optionally enable/hide interrupts from lower privilege modes.  This would allow the 2-stage translation and other features of the hypervisor to be used in a CLIC-based system, but would not support sending interrupts directly to descheduled guest OS contexts (have to route via hypervisor interrupt). 2) support multiple descheduled CLIC contexts to which interrupts could be sent directly while they were sleeping, which would be a much bigger project. 3) add an extension to AIA (instead of working on CLIC) to somehow reduce interrupt latency using the standard hypervisor stack.

christian-herber-nxp, 2025-05-05:
closing as tracked in https://github.com/riscv/riscv-fast-interrupt/issues/248

closed 2025-05-05 by christian-herber-nxp