riscv/riscv-fast-interrupt #93: the content of the output port "level" of the clic connected to the core

gxflying, 2020-08-20:
Hi sir, when I learn the clic spec, I find it is NOT described whether the priority bits in **cintctl**  should be included in the "level" port which will be connected to a risc-v core?

if NOT ， then the **level port** will only contents the level bits  of  **cintctl**  with prioryty bits setting to all ones,   is it right?

hope for your reply, thanks
https://github.com/riscv/riscv-fast-interrupt/blob/master/clic.adoc#321-specifying-interrupt-privilege-mode

Kevin-Andes, 2020-09-08:
To save space, both interrupt "level" are "priority" are combined and packed in the 8-bit "clicintctl" register. The 8-bit interrupt level is used to determine preemption. In contrast, the 8-bit interrupt priority is only used as tie-breaker when there are multiple pending interrupts with the same interrupt level. Both level and priority are used by the RISC-V core.

Although level and priority are both 8-bit unsigned values, the actual bits implemented can be fewer than 8 (and they can be determined by clicinfo.CLICINTCTLBITS and cliccfg.nlbits). If the actual bits implemented are less than 8, then they left-justified and appended with 1's for the lower unimplemented bits.

Thanks for asking and I will further clarify the spec based on the question you brought up.

https://github.com/riscv/riscv-fast-interrupt/blob/master/clic.adoc#321-specifying-interrupt-privilege-mode
CONFIDENTIALITY NOTICE:

This e-mail (and its attachments) may contain confidential and legally privileged information or information protected from disclosure. If you are not the intended recipient, you are hereby notified that any disclosure, copying, distribution, or use of the information contained herein is strictly prohibited. In this case, please immediately notify the sender by return e-mail, delete the message (and any accompanying documents) and destroy all printed hard copies. Thank you for your cooperation.

Copyright ANDES TECHNOLOGY CORPORATION - All Rights Reserved.

Kevin-Andes, 2020-10-20:
Clarified in the spec with the following commits:
ad46f39d163057daabf190193fb767716dea8326
10308986924346b17cd41042989457508f2955d1

closed 2020-10-20 by Kevin-Andes