riscv/riscv-fast-interrupt #95: Remove N - User Mode Interrupt from doccument

David-Horner, 2020-08-27:
N extension is likely to replaced. S mode with base address space appears to be a better near term alternative.
Removing references of it from the document will make it more the remainder understandable.
It will also free up some bits in the xnxti registers as only MIE and SIE remain relevant.

kasanovic, 2020-09-08:
Given that N has not been ratified, and is not being pushed at this time, it makes sense to drop references to N mode from the CLIC spec.  I would not want to repurpose bits that might be used for user-level interrupt handling at this time however.

kasanovic, 2021-02-17:
Closed by https://github.com/riscv/riscv-fast-interrupt/commit/de1c53f3eb7da83951bf8d003e962ca4d07c240b

closed 2021-02-17 by kasanovic