riscv/riscv-fast-interrupt #97: Proposed reformat of *cause CSRs

kasanovic, 2020-09-08:
From John Hauser, 14 March 2019:
The overloading of the '*cause' CSRs with new fields has always felt
awkward to me, even though I accept it was well motivated by a desire
to keep the instruction counts to a minimum.  Fortunately, I believe
I've found a different arrangement, just as good, that avoids CLIC mode
needing to modify the '*cause' registers.

In my new proposal, the complete set of CSRs that would be added with
CLIC are these:

   0xm07   *tvt           Trap-handler vector table base address
   0xm45   *istatus       Interrupt status
   0xm46   *maskil        Masked interrupt level
   0xm47   *nexti         Interrupt handler address and enable modifier
   0xm48   *scratchcsw    Conditional scratch swap on priv mode change
   0xm49   *scratchcswl   Conditional scratch swap on level change

The '*tvt', '*nexti', '*scratchcsw', and '*scratchcswl' CSRs are all
unchanged from before (aside from my adjusting the name and encoding of
'*nxti').  I'm not sure about the exact encodings, but I've mostly kept
them the same as before.  The real changes are all in '*istatus', which
coalesces fields that were previously in '*cause' and '*intstatus'.
I also went ahead and added '*maskil' for setting an interrupt mask
level, something that was already on the group's agenda.

To keep it simple, I'll usually just use M mode as an example.  Here is
the proposed format for 'mistatus':

mistatus

   XLEN-1   TT              trap type: 0 = exception, 1 = interrupt
    27:26   MPP             alias of mstatus.MPP
       25   MPIE            alias of mstatus.MPIE
       24   PTT             previous trap type
    23:16   PIL             previous interrupt level
     15:8   IL (read-only)  current interrupt level
      7:5   (reserved)
        4   HV              1 = reading trap vector table (was 'inhv')
      3:0   (reserved)

You'll recognize most of these fields from the earlier 'mcause'.

The new 'mmaskil' is a read-write 8-bit register.  New interrupts
can be taken in M mode only for an interrupt level greater than
max('mistatus.IL', 'mmaskil').  There are no limitations on writing
'mmaskil' from M mode.  On initial entry to an interrupt trap, it
is guaranteed that 'mmaskil' < 'mistatus.IL', because otherwise the
interrupt couldn't have been taken.  Code that wants to raise the mask
level (typically an interrupt handler, but could be anything) should
usually save the old value first and then later restore it.

In CLIC mode, the hardware performs these extra actions on any trap
entry in M mode:

    mistatus.PTT = mistatus.TT
    if (trap is an interrupt) {
        mistatus.TT = 1
        mistatus.PIL = mistatus.IL
        mistatus.IL = new interrupt level
    } else {
        // trap is a synchronous exception
        mistatus.TT = 0
    }

These actions are added on MRET:

    if (mistatus.TT) {
        mistatus.IL = mistatus.PIL
    }
    mistatus.TT = mistatus.PTT
    mistatus.PTT = 0

An interrupt handler that enables nested interrupts needs to save and
restore 'mistatus' instead of 'mcause'.  Note that on entry to a trap
in M mode, the sign bits of 'mcause' and 'mistatus' will be the same
(although this relationship does not maintain forever).

One advantage of the new scheme is that, because the '*cause' registers
are left unchanged, synchronous exception handlers can be far more
oblivious to CLIC mode than with the current specification.  I've
arranged things so that, if an exception handler wants to save state
and re-enable interrupts to allow for nested traps, it can do so in
the same way it would without any knowledge of CLIC mode.  (Why it
would want to do so is a valid question, but I see no reason to exclude
the possibility gratuitously.)

I believe all the code examples in the CLIC document work the same in
the new scheme, assuming 'mcause' is replaced by 'mistatus'.  For the
interrupt-driven C-ABI example, the sequence

   li a0, (MMODE)<<PP|(0)<<PIL|(1)<<PIE
   csrw mcause, a0      # Initialize mcause to have pp=M, pil=0, pie=1

must be replaced by this equivalent code:

   li a0, (1)<<TT|(MMODE)<<PP|(1)<<PIE|(0)<<PIL
   csrw mistatus, a0    # Init mistatus to have TT=1, MPP=M, MPIE=1, PIL=0

kasanovic, 2020-09-08:
From Kevin Chen:
Please note that the LSB of "mcause" contains the interrupt ID (12-bit        
exccode) which is automatically updated by HW when a new interrupt is taken.  
If you don't back up mcause, this interrupt ID will be overwritten by more    
critical interrupts in the case of nesting interrupts. Once it is overwritten 
(and gone) without backup, the ISR will miss this piece of information and may
not be able to complete successfully.

kasanovic, 2020-09-08:
From John Hauser:
Except that none of the examples in the CLIC document actually looks at
the bottom 12 bits of *cause for handling interrupts.  When interrupts
are hardware-vectored, there is usually no need for an interrupt
handler to look at the interrupt number in *cause.  But even when
interrupts are not hardware-vectored, so long as we set up a table of
interrupt handlers pointed to by *tvt, the special *nexti CSR gives
us a pointer into this table without making any use of the interrupt
number in *cause.  If I'm not mistaken, all the examples in the
CLIC document use one of these two methods, which is why I'm able to
say that moving the fields out of *cause hasn't changed any of the
examples, other than replacing *cause with my new *istatus.

Futhermore, let's not lose sight of the fact that the current CLIC
already proposes techniques using *nexti that don't reliably maintain
the interrupt number in *cause.  I call your attention to the "Example
Unix C ABI interrupt trampoline" in section 7.1.  That code has two
instances of

    csrrsi a0,mnxti,MIE

which may claim a new pending interrupt, causing the interrupt number
in mcause to be updated.  But in each case, the updated value of mcause
could be wiped out immediately afterward by a higher-level interrupt,
because interrupts are enabled.  So the value that was stored in mcause
can't be relied on by an interrupt handler.  The only reliable
indication of the interrupt identity is the value written to register
a0.  My proposal wouldn't change that.

The same thing happens again with the example for the "interrupt-driven
C-ABI model" in section 8.  A higher-level interrupt that occurs
immediately after the access to mnexti will wipe out the information in
mcause.  (Of course, it also happens again when the trampoline example
from above gets expanded in sections 9.2 and 10.1.)

If interrupt handlers don't use the special methods supported by CLIC
for vectoring interrupts and instead depend on the interrupt number in
*cause, they will be a little slower.  But even then, it's not clear to
me that my proposal changes things by much.  Can you provide example
code for the sort of situation you're concerned about?

kasanovic, 2020-09-08:
From John Hauser, 19 March 2019:
I've realized I can make my proposal a bit simpler.  And by "a bit"
in this case I literally mean one flip-flop.  The *istatus CSR in my
earlier proposal has a bit "PTT" for "previous trap type".  I once had
a reason for needing this bit, but the reason went away; it's no longer
needed.  So here's the new proposed format for mistatus without PTT:

mistatus

   XLEN-1   TT              trap type: 0 = exception, 1 = interrupt
    26:25   MPP             alias of mstatus.MPP
       24   MPIE            alias of mstatus.MPIE
    23:16   PIL             previous interrupt level
     15:8   IL (read-only)  current interrupt level
      7:5   (reserved)
        4   HV              1 = reading trap vector table (was 'inhv')
      3:0   (reserved)

On a trap entry to M mode, when CLIC mode is enabled:

    if (trap is an interrupt) {
        mistatus.TT = 1
        mistatus.PIL = mistatus.IL
        mistatus.IL = new interrupt level
    } else {
        // trap is a synchronous exception
        mistatus.TT = 0
    }

On MRET:

    if (mistatus.TT) {
        mistatus.IL = mistatus.PIL
    }
    mistatus.TT = 0

Naturally, S mode and U mode would have their own sistatus and
uistatus, with equivalent behavior.

As before, if an interrupt handler wants to re-enable interrupts to
allow nested higher-level interrupts, it must save and restore *istatus.
However, synchronous exception handlers generally do not need to know
about *istatus, and they also see no change in the *cause CSR, because
my proposal leaves it the same whether CLIC mode is enabled or not.

kasanovic, 2020-09-08:
Related to #82

David-Horner, 2020-09-17:
Whether or not we use this proposed structure, we should seriously consider leaving *cause alone and a new register designation be used.
the privileged 1.1 was ratified with a definition of *cause that is not extensible.
Had it defined a range of bits from 30 through 12 as reserved (tactically WPRI) we could have used that space without (meaningful) controversy..
However, it did not and at best we can argue that the single field Exception Code is sufficiently under defined to allow our extended use.
This position is clearly challenged by the ratified definition:
> The mcause register is an MXLEN-bit read-write register formatted as shown in Figure 3.25. When
> a trap is taken into M-mode, mcause is written with a code indicating the event that caused the
> trap. **Otherwise, mcause is never written by the implementation**, though it may be explicitly
> written by software.
> The Interrupt bit in the mcause register is set if the trap was caused by an interrupt. The Exception
> **Code field contains a code identifying the last exception. Table 3.6 _lists the possible_ machine-level
> exception codes.** The Exception Code is a WLRL field, so is only guaranteed to hold supported
> exception codes.

I believe it possible to extend the allowed exception codes. It appears implicit that more codes may need to be added.
But I do not believe subdividing the field can be done with integrity of the implicit guarantee that the Foundation holds backward compatibility as an extremely high goal, breaching only when failing to do so would be irresponsible to the ecosystem, not solely to our members.   

There is a lesson to be learned here, that appears to be applied to our current definitions of reserving unused space.
However, equally important is attempting to leave this reserved space usable. e.g. although mintstatus reserves space, the prime real estate of bit 31 (sign) is not available for expanding to a significant signal for state information, rather it is used by an 8 bit field that has no special significance for its sign bit.

kasanovic, 2020-09-22:
In TG meeting, 9/22/2020 - adding these fields to mcause does not  really affect backwards-compatibility, as the upper bits of the interrupt cause values were designated for platform use.  The higher values of exception cause were also reserved.  The text of the privileged ISA document should be clarified.

billhuffman, 2021-11-09:
Several comments on this thread:
- There seems to be the assumption that enabling interrupts during an exception handler (and maybe even during an interrupt handler) would be a somewhat unusual situation.  In the embedded world, for which this spec is written, the number of cycles for which interrupts are disabled is an important parameter.  Less than 10 cycles is desirable.  That means interrupts are re-enabled almost immediately on every exception.
- There also seems to be an assumption that, when hardware vectoring is used, the interrupt number is clear simply from the handler that is running.  But it is common to have multiple interrupts have identical entries in the vector table and then, in a small number of locations, do different things depending on which interrupt was taken.
- Perhaps a0 can be used to determine the interrupt number by subtracting xtvt and shifting right 2 or 3 bits as the standard mechanism, but I'm not at all sure that's the current expectation.
- I think that the spec has the assumption that the low bits of xcause can be used to determine the interrupt being processed.  But there seems to be a hole in this in that, when hardware sets the interrupt number at the time an interrupt is taken, xIE is always cleared.  And the code saves xcause before setting xIE again.  On the other hand, when xnxti is used for a very similar purpose, xIE is usually set - leaving, as John Hauser points out, a cause value that can be overwritten.  I think we do need to decide which set of assumptions is correct from the point of view of the handler.

dansmathers, 2021-12-16:
I think in the cases that John Houser points out, level interrupts work fine, but not edge interrupts.  With level interrupts, the interrupt source isn't cleared with xnxti and the lower priority interrupt is still pending.  So it's okay for xcause to be overwritten because the interrupt will just be handled later when it is the highest priority.  It is a good point so that should be clarified in the doc.  I think the key usage for using  edge triggered clearing with xnxti would be later in the example where MIE is cleared instead of enabled to allow SW tailchaining.
csrrci a0, mnxti, MIE   # Claim highest current interrupt. (16)

kasanovic, 2022-01-04:
Discussed in TG meeting.  It is not clear that exception handlers can be oblivious to CLIC and work correctly in CLIC mode.  Providing a separate `xcliccause` CSR that shadows `xcause` and `xstatus` CSR was discussed but it wasn't clear what advantage this provides over current scheme.   The `xintstatus` fields are read-only, primarily for debug, so do not need to be included there.

closed 2022-04-26 by kasanovic