riscv/riscv-fast-interrupt #98: Name of the controller: CLIC or FLIC?

kasanovic, 2020-09-08:
It is suggested that CLIC and CLINT are too close in name and sound.

The CLINT name has propagated and is being used widely for the original scheme, despite not having a clear definition anywhere.

It is proposed we should therefore change the name of CLIC to something that is further lexically from CLINT.  One proposal is FLIC for "Fast Local Interrupt Controller".

kasanovic, 2020-09-08:
CLIC name has been around for a while, and I don't think it really needs changing, and that CLINT and CLIC are reasonably distinguishable.

Kevin-Andes, 2020-09-17:
The main problem is that these 2 names are NOT self-explanatory. Also, by having identical words and meaning, they are too confusing for millions of new readers in the future:

   Core Local INTerruptor (for CLINT)
   Core Local Interrupt Controller (for CLIC)

Therefore, before ratification, we better replace them with clear self-explanatory names to avoid confusion and misuse in the future.

billhuffman, 2020-09-24:
I'm not eager to change, but a potential name for CLIC is CNIC, Core Nested Interrupt Controller.   That contains its biggest differentiation.

kasanovic, 2021-08-31:
We're agreeing to sticking to CLIC as that's what we have been using.

closed 2021-08-31 by kasanovic