riscv/riscv-fast-interrupt #100: reserve immediate bits 0, 2 and 4 in csrrsi rd, xnxti, uimm[4:0] and csrrci rd, xnxti, uimm[4:0]

David-Horner, 2020-09-14:
Bits 0, 2 and 4 in xstatus are currently WPRI (Write Preserve;Read Ignore).
They are essentially reserved in xstatus for future use.
We cannot know how these will be repurposed.
Specifically, they may be repurposed to have no value for Fast-Interrupt use.
If these bits are reserved, xnxtir can evolve with meaningful use for these 3 bits, potentially unconnected with xstatus..
Note:That the majority, 3 of the 5 immediate bits, are reserved.

The pseudo code for csrrsi rd, mnxti, uimm[4:0] becomes:
mstatus |= (uimm[4:0] & 0b01010); // Performed regardless of interrupt readiness.
and for csrrci:
mstatus &= ~(uimm[4:0] & 0b01010); // Performed regardless of interrupt readiness.

kasanovic, 2022-01-18:
We will constrain the currently defined encodings for using the mnxti register.  This has the benefit of informing software of the canonical way to encode accesses to current mnxti. This does not presume a future use for those bits, as in future specification we might use them as in the regular csrrsi/csrrci behavior.

kasanovic, 2022-08-02:
Closed with https://github.com/riscv/riscv-fast-interrupt/commit/13fa19e874f7b526b57dda0160e7ee3f21df5235

closed 2022-08-02 by kasanovic