riscv/riscv-fast-interrupt #101: Allow xnxti to trigger on equal level as saved interrupt context.

David-Horner, 2020-09-14:
The spec currently reads:

>The xnxti CSR can be used by software to service the next horizontal interrupt for the same
privilege mode **when it has greater level than the saved interrupt context (held in xcause`.pil`)** and
greater level than the interrupt threshold of the corresponding privilege mode, without incurring
the full cost of an interrupt pipeline flush and context save/restore.

I propose that xnxti execute on pending interrupts of the current pil level.
I believe we can accommodate this as a change in the fundamental behaviour.
That most systems do not need to exclude same pil level invocation. 
This can be accomplished by appropriately defining the individual interrupts pil levels.

However, if it is deemed desirable to maintain the current definition, 
I propose that a 1 in the (hopefully) now reserved lowest bit in csrrs[i] rs1 or imm[0] 
  allow equal as well as greater pending pil to action.

In the next revised 7.2 preemptive code example I will assume the low bit enabling will include equal to current pil processing.

David-Horner, 2020-09-16:
On reviewing chapter 8 examples I believe it best to leave xnxti,x0 with clic.level > mcause.pil  triggering.

As previously suggested,  use the least significant bit in rs1 or imm[0] to enable this new mode.
  Specifically, trigger when clic.level >= mcause.pil and also even when SHV is the selected highest pending interrupt.
If SHV, address is vector's address with least significant bit also set.
Thus the program can differentiate if it wishes, and if it does not need to, then when used as a jump it is ignored (per usual jump behaviour).

Still a definition in progress. I will do the full write up later, but wanted to advise on progress (and avoid the debate about changing current behaviour).

christian-herber-nxp, 2025-04-07:
nxti was removed

closed 2025-04-07 by christian-herber-nxp