riscv/riscv-fast-interrupt #105: is mcause inhv writeable?

dansmathers, 2020-09-29:
Is mcause inhv writeable so if firmware writes to the xcause CSR, the value will stay?   If firmware does set inhv to 1 and it stays, when is inhv updated to reflect the actual hardware vectoring status?  after interrupts and exceptions?  only if a hardware vector successfully completes and sets inhv to zero?
We are doing some randomized verification and writing values to xcause csrs and this situation showed up.

Thanks

Kevin-Andes, 2020-10-06:
Based on TG discussion on 20201006:

This bit is writable, so the value will stay if firmware writes to it. Normally, it is set and clear by HW state machine for vectored mode interrupts. It is not updated for exceptions. xret does not clear it (to maintain backward compatibility). When this bit is set, it enters exception but other parts of xcause does not get updated/cleared. 

Need to determine which of the following options works the best.  If HW fetch for vector table is not successful, the machine will execute the SW exception routine with mcause reflecting one of the following: 
1. mcause.interrupt = 1, exception code = vector number, inhv = 1
2. mcause.interrupt = 0, exception code = some existing fault,       inhv = 1
3. mcause.interrupt = 0, exception code = a new type (group) of interrupt fetch faults,   inhv = N/A 

As long as SW does not clear this interrupt, the same interrupt will trigger again after the fault is handled by SW.

Is this type fault even recoverable at all?

Can we ignore the vector lookup failure and fall back to pure SW ISR? Can we fall back to pure SW mode if HW vector mode fails? Do we need 2 versions of ISR in this case? Any corner cases when we have nesting interrupts?

Without vector number, it may be difficult to figure out which part of the vector table went wrong (could be captured in mtval).

Will check with Krste about his original intention.
 
Need to clarify the details in the spec.

kasanovic, 2020-10-20:
The inhv bits are only written by hardware during the table vector read operation.  The inhv bits can be written by software, including when hardware vectoring is not in effect. The inhv bit has no effect except when returning from interrupt.

kasanovic, 2020-10-20:
One use case for restart is when there is a bare-S mode running on top of a security M-mode monitor.  The M-monitor can take M-mode interrupts in the middle of an S-mode interrupt sequence - the inhv bit allows the hardware to preempt the S-mode vector table read to take the M-mode interrupt.  The M-mode interrupt handler does not have to be aware of the S-mode interrupt handler state and can simply mret as before, in which case the hardware will replay the table read.

An implementation could choose to make the interrupt/table-read sequence non-interruptible (at the cost of worst-case interrupt latency), so inhv is never reported.  But some mechanism is still needed to indicate that the table read itself caused a trap, even if fatal.   We use inhv to report this case, as the bit is used to support the other scenarios.  Using a separate cause field for this case would prevent restart in other cases.

The cases when the table read exceptions are not fatal traps are when this code is running in a lower-privilege mode under some higher-privilege mode that takes action on the table read trap then resumes (.e.g,  user-mode interrupts with S-mode paging, or S-mode guest OS on hypervisor, or U/S-mode code under M-mode that has access logging on turned on).

When the table read causes a horizontal trap, this usually signifies an error in the software configuration.  These horizontal table read traps are not generally recoverable, so some separate debug mechanism will usually be required to determine the cause of the error.  The table address will be available in mtval  for debugger to inspect.

billhuffman, 2020-10-20:
I think there are three possible mechanisms for handling a trap of any kind during a hardware vector table lookup.  Let's say the privilege level of the vectored interrupt that has its lookup in progress is b and the privilege of the taken trap is a.  The following three ways may be ways in which the trap could be handled:
1. The trap can be delayed until the vector table lookup completes and the lookup value placed in aepc (this only works if the trap is not caused by the vector table lookup).  The trap at privilege a will be handled and will return to privilege b to execute the handler at aepc.
2. The vector table lookup can be abandoned and the state of privilege level b left with the interrupt taken but the vector table lookup remaining to be done.  Then, bcause.inhv is set and the privilege a trap taken.  It's not clear what aepc should be set to.  On return, because the bcause.inhv bit is set, the return process is entirely different.  The table location is recomputed from bcause and looked up again.  When the lookup is successful, bcause.inhv is cleared and the handler entered.
3. The entire interrupt can be postponed.  For this, the b interrupt's ip bit must be left set and the entire status register and current privilege kept at (or set back to) what they were before the privilege b interrupt was taken.  Then aepc set to the current bepc, and status updated for the trap to privilege a as if it were from the original instruction.  The return will be to the original instruction being executed before the b privilege vectored interrupt and the b privilege vectored interrupt taken all over again.  Interrupts work smoothly for this mechanism.

An implementation may only want to use a single mechanism.  That leaves out mechanism 1 as it can only be used for interrupts (and latency may be a problem even there).  Mechanism 2 can be used for interrupts and vertical traps and can be straightforwardly recovered from for those, but cannot be used for horizontal traps.  Mechanism 3 may be able to recover from horizontal traps, but causes the software more difficulty in figuring out what happened even for vertical traps.  It has to notice that atval points to a vector table address and know that the trap has nothing at all to do with the code that was apparently executing at the time of the trap but was because of a vectored interrupt.  This may be possible, but needs to be understood architecturally.

Mechanism 2 works straightforwardly except that we define horizontal traps as unrecoverable where we define them to do something for debug purposes.  Mechanism 3 could be used for all cases, but would need to be considered architecturally so that software would be able to deal correctly with it.  The fact that mechanism 3 needs special trap consideration even for vertical traps may make mechanism 2 and fatal horizontal traps more attractive.

If we choose mechanism 2 and don't allow mechanism 3 (by not saying what the software must handle), we need to say what aepc is set to and we need to define what a horizontal trap does.

kasanovic, 2021-04-13:
Closed by PR #131

closed 2021-04-13 by kasanovic