riscv/riscv-fast-interrupt #106: enhance support co-routines and thus allow level change already saved registers.

David-Horner, 2020-10-05:
enhance support co-routines and thus allow level change already saved registers.
can be used horizontally
application can check that co-routine did not clobber the registers reserved for itself
one halt loop suffices for multiple levels
current code assumes level 0 and pause for horizontal interrupt?
riscv/riscv-fast-interrupt@2fc1965

scenarios :
1) define code section that can be re-runnable even if interrupt clobbers a specified set of registers
co-routine code
 <<enable interrupts>>
 set redoable flag on register set q1
 1l:
 reset redo flag for q1
 .... execute re-runable code segment  ....
 if redo flag for q1 is set branch to 1l
 reset redoable flag for q1
 <<disable interrupts>>

interrupt co-routine handler code
  save non-q1 registers that are used in handler
  if redoable flag clear save q1 registers
  ‘’’’ execute some interrupt handler code ‘’’’ 
  if redoable q1 flag was set on entry
    and q1 register set modified
     set redo flag for q1  
   if redoable q1 flag not set on entry 
     restore q1 saved registers
  restore non-q1 save registers
  return 

Uses include 
  a) reducing code segments that need interrupts disbled
  b) reducing interrupt save/restores 

Specifically this code is applicable to nested interrupt handlers.  
2) on return from interrupt handler allow saved register set to be reused if immediately followed by an interrupt.
In the examples below the necessary modifications to the standard process are defined.

Note:  The exact order relative to each other and the existing functionality is sub-optimal and would be incorrect for some implementations.  For example, in 2a reset of reusable q1 flag in handler prolog could be performed co-incident with the read of its state.

Note:  In practice, the discreet functionality will need to be reordered to avoid special cases in the hardware design.  For example,

This functionality requires hardware support.

     A) This functionality allows reuse of the register sets saved in the trap handler
             by the next interrupt at the same level if:
               i) xret does not complete, or 
               ii) not instruction is executed before the next interrupt is actioned.

          For 2a hardware support is required but is minimal.
	i)  two bits in CSRs, reusable q1 flag for each privilege level
            ii)  if an interrupt is imminent (pending and enabled).
               MRET cannot execute any instructions until interrupt is engaged.
            iii) reusable q1 flag is reset if, after  return, any instruction is executed.

	

   B) The functionality in 2b is enhanced over 2a to allow 
        M-mode and S-mode to share the others saved register sets.

      2B requires the same hardware support as 2a, and 
        additionally requires S-mode to clear 
        M-mode's reusable q1 flag

2a) horizontal interrupts with minimal hardware flag reset. 
<<modified interrupt handler epilog
   a) leaves x2 in xswap pointing to saved q1 register set
   b) set reusable q1 flag for current privilege level x
   c) execute modified xret >>
[... 200 lines cut ...]
 when set informs co-routine (including interrupts) that the partner process is      
     able to fully restore qq state 
   coroutine is thus able to use qq state with impunity iff it signals dorecovqq to the co-routine.
      partner routine must at the end of the recoverable qq  section check do recovqq and recover qq state.
   further, it will clear statrecqq to terminate the shared qq state code region
 
plan: a single csrrci xnxti instruction does both 
        current dorecovqq state is placed in setrecovqq bit location and
           recovqq is cleared if setrecovqq is set in immediate field.       
Bit dorecovqq needs to be set on return from co-routine if coroutine messed with qq state.

Dorecovqq could be offset from zero that allows C.JR to return right back to process.  

is cleared when recovqq is set by csrrsi.+ 
new branch instruction on low bits set.
 generally available functionality , but especially valuable in a minimal register 
    use case where either:
       no other register available (without spill) to mask specific bits, 
          or reload of value costly (as CSR read may be). 

Note:  for 2a some implementations may already have this functionality.
		Specifically, if the implementation already detects pending enable interrupt and
	a)  vectors to start of interrupt routine and
	b)  leaves xepc unchanged
	Then the low bit of xepc can be used as the hardware signal for M and S modes.

Kevin-Andes, 2021-02-02:
Attached the detailed description and diagram that David sent out to the TG discussion mailing list:

[Tracking updated registers.pdf](https://github.com/riscv/riscv-fast-interrupt/files/5909980/Tracking.updated.registers.pdf)

[ill#1c.pdf](https://github.com/riscv/riscv-fast-interrupt/files/5909985/ill.1c.pdf)

Kevin-Andes, 2021-02-02:
In the discussion of our Task Group meeting on 20210119, some member pointed out that the worst case scenario won't get improved with this approach. This is because, even though the "window" to avoid repeated saving/restoring of interrupt context has been extended, the worst case will occur when an interrupt arrives right after this extended window.

Also, some member are concerned about the extra HW cost. Alternatively, a pure soft approach may achieve similar goal by using compiler to reserve a fixed group of registers for interrupts or only save registers that are actually used.

Lastly, the fastest approach should be using pure HW-assisted context saving/restoring, and it is not clear how this proposed scheme can work with pure HW approach. Also, we may need to examine if there are any potential security issues in the usage model.

christian-herber-nxp, 2025-05-05:
closing as not planned

closed 2025-05-05 by christian-herber-nxp