riscv/riscv-fast-interrupt #109: Add arch string for CLIC

kito-cheng, 2020-12-08:
Does it possible to add something like `ziclic` for CLIC for arch string? it would be easier to track the version and check CLIC is used from the software and toolchain view.

Kevin-Andes, 2020-12-10:
Hi Kito,

For clarification, do you mean creating a new instruction set extension (like Zfinx)? Or just creating a new compiler flag?

kito-cheng, 2020-12-10:
>For clarification, do you mean creating a new instruction set extension (like Zfinx)? Or just creating a new compiler flag?

More like creating a new instruction set extension, it would have an unify way to manage version and check its existence.

The usage will like other ISA extension, -march=rv32gc_ziclic to enable that explicitly on software site, and it also leverage the existing scheme to manage/present the version, e.g. -march=rv32gc_ziclic0p9 means CLIC v0.9.

Both LLVM and GNU toolchain folks has discuss this before, and most toolchain guys are happen with this way, otherwise we might create lots of compiler option for each new interrupt scheme or set of new CSR.

kasanovic, 2021-01-05:
Yes. As part of the arch profile work, we are breaking up the privileged architecture features into named extensions to allow the supported features to be listed clearly.  The CLIC would fall into this category for RVM22, and should have appropriate named and versioned sub-extensions.  The earlier versions had multiple options, so would want to not just list v0.9 for example, but have finer-grain distinctions on supported features.

dansmathers, 2021-02-09:
linking issue #80

kasanovic, 2021-08-31:
This is a privileged extension, though has some lower-privilege-mode-accessible features.   Given that the key CSRs are in M-mode, it should probably be named something like "Smclic".

kasanovic, 2021-12-21:
Closed with #184

closed 2021-12-21 by kasanovic