riscv/riscv-fast-interrupt #110: Insecure examples in the docs

bog-dan-ro, 2020-12-28:
While I was reading the `clic.pdf` doc I found some insecure examples (7.2., 8.1.).
In those examples you're using the `SP` register to save some other registers:
```
addi sp, sp, -FRAMESIZE # Create a frame on stack.
sw a0, OFFSET(sp) # Save working register.
# ....
```
which is very insecure as the `SP` register is set by the user application which can set it to any value.

aswaterman, 2020-12-28:
If you read the text, you'll see that one of CLIC's use cases is to support single-privilege-mode operation, for embedded use cases where the ISR and the task have the same privilege.  For those cases, continuing on the same stack frame does not pose a security problem.

Another use case is to support taking interrupts into higher privilege modes, in which case multiple stack pointers are used.  Search the text for xscratch.

bog-dan-ro, 2020-12-29:
Ah,somehow I skipped the 7.1 part :) , sorry for the noise.

closed 2020-12-29 by bog-dan-ro