riscv/riscv-fast-interrupt #111: Strange behaviour for faults accessing trap vector table

jrtc27, 2021-01-11:
The spec contains the following "interesting" sentence:

> If there is an access exception on the table load, xepc holds the faulting address. If this was a page fault, the table load can be resumed by returning with xepc pointing to the table entry and the trap handler mode bit set.

This is a strange case for software to have to handle, suddenly having to complicate its page fault handler for this special case. Why not just put the faulting address in xtval like a normal fault? Putting a data pointer in a code pointer register seems to be an abuse of the available registers, and will prove particularly awkward for CHERI where pointers are capabilities with associated permissions (and the `xepc` registers are extended to special capability registers `xepcc`, with `xepc` aliasing the integer address part); there's probably a way to make it work, but it'd be a bit gross to put a data capability in what should be a code capability register.

There should also be no need to special case the exception return; if the interrupt is still pending it'll be immediately taken again on xret, and if not then it no longer needs servicing so it's fine to not trap.

That's also, incidentally, the effective behaviour you'd get if you didn't try to do anything special to support nested traps.

With those addressed it xinhv becomes unnecessary and can be removed. If desired, two new exception codes could be allocated for "load trap vector (access|page) fault", which is a far more space-efficient encoding.

I also don't see why all the xstatus bits need mirroring in xcause. Any decent OS is going to need to save/restore xcause anyway because that's where the actual hardware-affecting state is that it needs to preserve. And for the tiniest of embedded OSes it's really not going to make a performance difference asking them to look at a single extra XLEN-bit CSR. Putting all this stuff in xcause feels like another abuse, and xpil similarly belongs in xstatus or some other CLIC-specific CSR that's not xcause.

That should all simplify and clean up the spec IMO.

dansmathers, 2021-01-11:
FYI, During the Fast Interrupts TG meeting discussion of #105,  we also discussed thoughts on use cases for restarting after a page table fault during hardware vectoring.   Just want to link those discussions here.

In some implementations CSR accesses can be time-expensive due to their atomic nature and how they can affect other machine state.  That's one advantage of mirroring xstatus bits within xcause.    Does this mirroring cause issues for CHERI implementations or is your suggestion more of a simplification argument?

jrtc27, 2021-01-11:
On high-performance architectures you'll be running a full-fat OS where you will not notice an extra pipeline flush from a CSR access, plus reads at least will be increasingly pipelined (I believe the new SiFive U7 series does this?). On simple microcontroller architectures that's just a few cycles so really does not matter. Has anyone actually done any scientific analysis to justify the proposed changes? If it does turn out to make a significant difference in real use cases then fine, but I do feel it should be grounded in evidence not speculation.

No, the mirroring into xcause doesn't cause issues for CHERI, I'm just looking at it with an OS kernel developer hat on sighing at having two different ways of doing things for what doesn't seem like a particularly good reason from my perspective (and also the nuisance it'll be if we ever want to provide a CLIC in our cores).

jrtc27, 2021-01-11:
Also, on reflection, I do not think there is any real situation where you'd legitimately expect to fault on accessing the trap table.

Access faults should never happen and are a sign you've done something very wrong.

Page faults could be that you've screwed up your page tables, in which case that's also wrong. Or you decided to swap out your trap table, but you're unlikely to be able to swap it back in without taking an interrupt; the I/O subsystem is going to be interrupt-based in a real OS so would need a special polled I/O path in _every single driver_, and will want to sleep and switch to another thread whilst the request completes, but you cannot do anything useful with interrupts disabled. So I cannot see how you would ever legitimately end up with a page fault. Plus, the table is tiny (especially in comparison to page tables), there's really no need to page it out.

So IMO both would be treated as fatal panics if they ever occurred in an OS, and my proposal still lets you (a) distinguish them so you know to panic (b) if you really really really want to adapt your OS to support such a very niche use case then you still can, but no hardware complexity is wasted on trying to support that efficiently.

dansmathers, 2021-02-09:
linking open issue #26

dansmathers, 2021-02-09:
lalso inking open issue #46

dansmathers, 2021-02-09:
also linking open issue #87

kasanovic, 2021-02-16:
There are two separate issues here - packing of other bits into cause is already recorded as #97, so please continue that discussion on that issue.

Any horizontal trap in an interrupt handler is non-resumable until exception state has been saved.  Using either mtval or xepc does not change this (#87).

The reason to support resumable faults on vector tables is in case a more privileged level has virtualized/emulated the RTOS+application using the CLIC, and could have paged out the vector table (or just wants to track accesses to it).

I believe the original motivation to use xepc instead of mtval was that the original CLIC proposal used instruction fetch instead of data fetch to access the vector table, and effectively switched to a new ISA for vector table fetches which only had a single instruction: "jump to address encoded in instruction".  This trick avoided need to route memory load data to PC, which is a path not otherwise present in processors, and also allowed restart to be handled as an instruction fetch restart.  Using xepc instead of mtval avoided the need for new hardware path from mtval to pc.

It would be more consistent to use mtval to hold the faulting table address, as this would match the existing privilege architecture spec.

jrtc27, 2021-02-16:
> There are two separate issues here - packing of other bits into cause is already recorded as #97, so please continue that discussion on that issue.
>
> Any horizontal trap in an interrupt handler is [...]

That's fine, the higher privilege level can take the fault, page the data back in (or just do its bookkeeping), xret and then let the still-pending interrupt trigger again? To the guest it just looks like a very slow page table walk, unless the interrupt stops being pending, at which point well you don't actually need to take it any more so it's fine not to (and if missing those interrupts is a problem, well, you already have races, this just makes them more likely, so either way you should make them edge-triggered).

kasanovic, 2021-02-17:
I talked over this point with Andrew, to help reconstruct the original thinking, which predates many subsequent changes to the spec.  Our conclusion was that now both xtval and xepc should be written with the faulting vector table address.

The interesting case is vertical traps, where a more privileged layer is handling page faults or other synchronous faults on vector table access.  The regular code path in more privileged layer will want to use xtval to determine what bad virtual address to page in, but will not normally restore xtval when returning to faulting context (potentially after some time and other contexts have run)  However, it will restore xepc  (using x for more privileged mode here) before using xret on normal code path.

Hardware would detect the inhv bit being set during xret and resume table load using value in xepc (or generate instruction fetch with "jump-to-absolute-address" ISA in place).

I don't believe in general we can avoid resuming table load (versus restart of original interrupted instruction).    Implementations will be simpler if they can treat the interrupt vectoring in two steps (or even using sequence of instruction fetches in one possible implementation style), instead of a single-step instruction which has a lot of CSR updates to back out of on fault.  Some interrupts can be edge-triggered and would be lost under restart model. Restart would also require both original code and handler can be co-resident in TLBs for progress (should be straightforward, but is an additional constraint).  We'd also have to change the xcause reported to distinguish regular page faults on the interrupted instruction versus interrupt vector table faults.   There is also some potential confusion with different privilege modes for interrupted instruction and table reads.

So proposal is to add xtval being written  (which also would be consistent with current privilege spec) along with xepc.

kasanovic, 2021-02-17:
I think this matches reasoning in #105, with option 2 being the surviving option.

kasanovic, 2021-04-13:
Closed by PR #131

closed 2021-04-13 by kasanovic