riscv/riscv-fast-interrupt #117: Section 3.2.1:  cliccfg.nmbits = 1 with M/S/U modes supported

konrad-schwarz, 2021-02-26:
The text says that in a system supporting M, S, and U modes, with nmbits set to one, msb clear results in the interrupt taken in U mode.  However, the table below shows the interrupt being taken in S-mode -- a much more logical and useful settting, IMHO.

Kevin-Andes, 2021-03-02:
Thank you for pointing out this inconsistency. According to the privilege spec (Table 1.2 in Sec. 1.2, supported combinations of privilege modes), it describes the supported modes are M and U if the number of privilege levels supported are 2. In this case, there is only 1 mode bits so the effective levels are 2.

We can discuss further which combinations are more common use cases: M plus U, or M plus S.

konrad-schwarz, 2021-03-03:
Unfortunately, I have scheduling conflicts during your meetings.

However, I’m not sure there is much to discuss here:

On a system supporting just M and U mode – which at least originally was the micro-controller mode, it is obvious that these two modes should be used.

On a system supporting M, S, and U mode – the standard “application processor” mode, required for running an operating system such as Linux, the ability to direct interrupts to S-mode – the mode the operating system is running on – is a non-negotiable requirement (otherwise interrupt driven IO is not possible).   The need for M-mode to handle interrupts, e.g., temperature warnings, fan settings, battery management, and so on, is well established industry practice; the analogue to M-mode on Intel x86 for example is System Management Mode (SMM) which uses a dedicated System Management Interrupt that has higher priority than any ordinary interrupt, and rightfully so, as ignoring or delaying handling of these interrupt requests can imperil the machine.

Kevin-Andes, 2021-03-04:
No problem, feel free to put down your comments/suggestions here and I can later bring them up in the meeting for discussion.

I see your points. For clarification, does Linux **only** need M-mode and S-mode interrupts typically? Or does Linux also need U-mode interrupts? If it needs all 3 modes then 2 bits must be used for privilege selection.

By the way, as an example, a potential use case of just M and U modes is when you use this processor as a micro-controller (even though it has capability of supporting M/S/U modes).

dansmathers, 2021-04-13:
This line is for a system has M, S, and U mode so if there is only 1 bit for nmbits, then it should select between M and S mode.

kasanovic, 2021-04-13:
Agree with @dansmathers - the table is correct. When the system has all three M,S,U modes, then a single bit should select between M and S.  Bare S mode (no MMU, satp=0) can be used instead of the proposed N-extension to delegate interrupts out of a secure M mode in an embedded Microcontroller.  Bare S mode is already ratified as part of privileged architecture.

kasanovic, 2021-04-13:
Closed by pull #132

closed 2021-04-13 by kasanovic